dusty-nv/jetson-inference #619: DEV Branch | [OpenGL] failed to create X11 Window. |  jetson.utils -- no output streams, creating fake null output

Mohit-Ak, 2020-06-19:
**Branch** : **DEV**
**Issue**
There is a ```[OpenGL] failed to create X11 Window.``` when ```import jetson.inference``` is used before  the output statement ```output = jetson.utils.videoOutput(output_URI, argv=sys.argv)```

**STEPS TO REPRODUCE**

1. ```git clone -b dev --recursive https://github.com/dusty-nv/jetson-inference```
2. Compile the library using the steps mentioned [here](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md). Ensure that you use "**dev**" branch.
2. Create and run the following piece of python code.
**CODE**
```
import jetson.utils
import sys
import jetson.inference
input_URI="rtsp://192.168.1.2/axis-media/media.amp?videocodec=jpeg&camera=1&size=1&camera=1"
output_URI=""

# create video sources & outputs
input = jetson.utils.videoSource(input_URI, argv=sys.argv)
output = jetson.utils.videoOutput(output_URI, argv=sys.argv)

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
input_URI="rtsp://192.168.1.2/axis-media/media.amp?videocodec=jpeg&camera=1&size=1&camera=1"
output_URI=""
# capture frames until user exits
while output.IsStreaming():
	image = input.Capture()
	detections = net.Detect(image, image.width, image.height)
	output.Render(image)
	output.SetStatus("Video Viewer | {:d}x{:d} | {:.1f} FPS".format(image.width, image.height, output.GetFrameRate()))
	
```
**ERROR**
```
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for 192.168.1.2/axis-media/media.amp?videocodec=jpeg&camera=1&size=1&camera=1
[gstreamer] gstDecoder -- discovered video resolution: 960x720  (framerate 30.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  image/jpeg, framerate=(fraction)30/1, width=(int)960, height=(int)720
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://192.168.1.2/axis-media/media.amp?videocodec=jpeg&camera=1&size=1&camera=1 ! queue ! rtpjpegdepay ! nvjpegdec ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from rtsp://192.168.1.2/axis-media/media.amp?videocodec=jpeg&camera=1&size=1&camera=1
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: rtsp://192.168.1.2/axis-media/media.amp?videocodec=jpeg&camera=1&size=1&camera=1
     - protocol:  rtsp
     - path:      192.168.1.2/axis-media/media.amp?videocodec=jpeg&camera=1&size=1&camera=1
     - extension: 
     - port:      -1
  -- deviceType: ip
  -- ioType:     input
  -- codec:      mjpeg
  -- width:      960
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   false
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  3840x2160
[OpenGL] glDisplay -- X window resolution:    3840x2160
[OpenGL] failed to create X11 Window.
jetson.utils -- no output streams, creating fake null output
jetson.inference -- detectNet loading build-in network 'ssd-mobilenet-v2'
detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

```

Mohit-Ak, 2020-06-19:
**Temporary Solution**

Import the ```jetson.inference``` after the output statement.
That is, after ```output = jetson.utils.videoOutput(output_URI, argv=sys.argv)```

**Working Code**

```
import jetson.utils
import sys
input_URI="rtsp://192.168.1.2/axis-media/media.amp?videocodec=jpeg&camera=1&size=1&camera=1"
output_URI=""
#!/usr/bin/python3
# create video sources & outputs
input = jetson.utils.videoSource(input_URI, argv=sys.argv)
output = jetson.utils.videoOutput(output_URI, argv=sys.argv)

import jetson.inference
net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
input_URI="rtsp://192.168.1.2/axis-media/media.amp?videocodec=jpeg&camera=1&size=1&camera=1"
output_URI=""
# capture frames until user exits
while output.IsStreaming():
	image = input.Capture()
	detections = net.Detect(image, image.width, image.height)
	output.Render(image)
	output.SetStatus("Video Viewer | {:d}x{:d} | {:.1f} FPS".format(image.width, image.height, output.GetFrameRate()))
	

```

dusty-nv, 2020-06-19:
What happens if you only `import jetson.inference` at the top, and no `import jetson.utils`?

mmatesic01, 2020-07-22:
> What happens if you only `import jetson.inference` at the top, and no `import jetson.utils`?

It gives the same error. At least for me. So I am using "Temporary Solution".

tanueihorng, 2020-08-04:
> What happens if you only `import jetson.inference` at the top, and no `import jetson.utils`?

Hi @dusty-nv 
The following code works well ! (encountered the same issue) 
`import jetson.utils

camera = jetson.utils.videoSource("rtsp://root:1234@10.50.14.15:18888/h264_ulaw.sdp", argv=['--input-codec=h264'])
display = jetson.utils.videoOutput("display://0") # 'my_video.mp4' for file

import jetson.inference
net = jetson.inference.detectNet("ssd-inception-v2", threshold=0.4)

while display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	display.Render(img)
	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))`

jc508, 2022-01-16:
Hi - some more evidence for this.
The 'swap' of videoSource and videoOutput did not work for me.
There are 2 complete programs here 
cv2-lesson50_1_pure_debug1.py  works and using jetson.utils.videoSource and jetson.utils.videoOutput
cv2-lesson50_3_mixed_debug1.py fails, is trying to use cv2 for camera input then jetson.utils.videoOutput for output

The failure is 
**[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1280x720
[OpenGL] failed to create X11 Window.
jetson.utils -- no output streams, creating fake null output**

In case it needed both videoSource and videoOutput I added a dummy camera then closed it before using cv2.VideoCapture
it didn't make any difference.
Using Nano and jetpack 4.6 with pre-installed Opencv 4.1.1

Hoping this allows you to reproduce the problem

working version
```
# using  jetson.utils.videoSource and jetson.utils.videoOutput

import jetson.inference
import jetson.utils
import time

width  = 1280     # 1640      # 1280
height = 720      # 1232      # 720
framerate = 50    # 30        # 60      # choose 30 or 60

param_videoSource = []
param_videoSource.append("--input-width=" + str(width))
param_videoSource.append(f"--input-height=" + str(height))
param_videoSource.append(f"--input-flip=vertical")
param_videoSource.append(f"--input-rate=" +str(framerate))

cam = jetson.utils.videoSource('csi://0',   argv=param_videoSource )   

## Using videoOutput
param_videoOutput = []
param_videoOutput.append("--width=" + str(width))
param_videoOutput.append(f"--height=" + str(height))

disp = jetson.utils.videoOutput('display://0', argv=param_videoOutput)
font = jetson.utils.cudaFont()

networkToUse = 'googlenet'
param_imageNet = []
param_imageNet.append("--network=" + networkToUse)
param_imageNet.append(f"--width=" + str(width))
param_imageNet.append(f"--height=" + str(height))
param_imageNet.append(f"--log-level=info")
net = jetson.inference.imageNet(networkToUse, argv=param_imageNet) 

disp.SetStatus("Network in use is '" + networkToUse + "'")

f = 0
while True:           
    f = f+ 1
    frame = cam.Capture()

    classID, confidence = net.Classify(frame, width, height) 
    item = net.GetClassDesc(classID)

    disp.Render(frame)
    
    # exit on input/output EOS
    if not cam.IsStreaming() or not disp.IsStreaming():
	    break

cam.Close()
disp.Close()
```

mixed mode one that fails
```
# uses cv2 for camera input then jetson.utils.videoOutput for output
import jetson.inference
import jetson.utils
import time
import cv2
import numpy as np

# for capture (ie camera) settings must pick 100% modes only eg 1640 x 1232, 1280 x 720, 1920 x 1080
capture_width  = 1280     # 1640      # 1280
capture_height = 720      # 1232      # 720
framerate = 50    # 30        # 60      # choose 30 or 60
# for display choose a corresponding multiple/fraction ie half 820x616, 640x360  or quarter 410x308/408x308 , 320x180
display_width  = 1280     # 1640      # 1280
display_height = 720     # 1232      # 720

## Using videoOutput
param_videoOutput = []
param_videoOutput.append("--width=" + str(display_width))
param_videoOutput.append(f"--height=" + str(display_height))

disp = jetson.utils.videoOutput('display://0', argv=param_videoOutput)      # OK
print("#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=")
font = jetson.utils.cudaFont()

my_gstreamer_pipeline_string = 'nvarguscamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)0/1 ! nvvidconv flip-method=6 ! video/x-raw, width=(int)1280, height=(int)720, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR !  appsink drop=True'
print('gstreamer_pipeline='  + my_gstreamer_pipeline_string )

cam=cv2.VideoCapture(my_gstreamer_pipeline_string , cv2.CAP_GSTREAMER)

networkToUse = 'googlenet'
param_imageNet = []
param_imageNet.append("--network=" + networkToUse)
param_imageNet.append(f"--width=" + str(display_width))
param_imageNet.append(f"--height=" + str(display_height))
param_imageNet.append(f"--log-level=info")
net = jetson.inference.imageNet(networkToUse, argv=param_imageNet)            # using a prepared list, again 1st param is ignored so list must include "--network="

disp.SetStatus("Network in use is '" + networkToUse + "'")

f = 0
while True:           
    f = f+ 1
    success, imageBGR = cam.read()
    imageRGBA=cv2.cvtColor(imageBGR,cv2.COLOR_BGR2RGBA).astype(np.float32) # improve this later
    imageCuda = jetson.utils.cudaFromNumpy(imageRGBA)

    if capture_width != display_width or capture_height != display_height:
        frame = jetson.utils.cudaAllocMapped(width=display_width,
                                       height=display_height ,
                                       format=imageCuda.format)
        jetson.utils.cudaResize(imageCuda, frame)
        if f == 1:
            print("Frame finally =" + str(frame))
    else:
        frame=imageCuda

    if f % 20 == 0:        # every so often print it as well otherwise there is no output because of videoOutput failure
        print("Frame # =" + str(f) + " returned classId=" + str(classID) )
    classID, confidence = net.Classify(frame, display_width, display_height) 
    item = net.GetClassDesc(classID)

    disp.Render(frame)
    
    # exit on input/output EOS
    if not disp.IsStreaming():
        break

cam.release()
disp.Close()

```

closed 2023-04-27 by dusty-nv