dusty-nv/jetson-inference #620: Where to adjust camera usage

glennford49, 2020-06-21:
Where can i adjust camera usage, i want to use usb camera as default before compiling

dusty-nv, 2020-06-21:
Are you using C++ or Python?

Instead of --camera command line string, pass "/dev/video0" (or whatever your v4l2 device is) to gstCamera initializer.

glennford49, 2020-06-24:
Im compiling it from this source , im running .py files  i want usb camera be the default so that i will not issue string commands /dev/video0 everytime i run python files,

dusty-nv, 2020-06-24:
Replace the command line string passed to gstCamera with yours, or in the argparse options set yours to be the default (that way you can still set it from the command line if you want to use a different camera, but by default it will use yours)