dusty-nv/jetson-inference #622: install-pytorch.sh ImportError: libcudart.so.10.0

DennisFaucher, 2020-06-23:
Might be a CUDA incompatibility. INstall looks for CUDA library 10.0 when CUDA 10,2 installed in latest Xavier SDK image. Is there a workaround? TIA.

$ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Wed_Oct_23_21:14:42_PDT_2019
Cuda compilation tools, release 10.2, V10.2.89

$ ./install-pytorch.sh
[jetson-inference]  Package selection status:  0
[jetson-inference]  Packages selected for download:  2
[jetson-inference]  Downloading PyTorch v1.1.0 (Python 3.6)...
[jetson-inference]  Checking for 'python3-pip' deb package...installed
[jetson-inference]  Checking for 'qtbase5-dev' deb package...installed
[jetson-inference]  Checking for 'libjpeg-dev' deb package...installed
[jetson-inference]  Checking for 'zlib1g-dev' deb package...installed
torch-1.1.0-cp36-cp36m-linux_ 100%[=================================================>] 202.36M  5.06MB/s    in 41s     
Processing ./torch-1.1.0-cp36-cp36m-linux_aarch64.whl
Collecting numpy (from torch==1.1.0)
Installing collected packages: numpy, torch
Successfully installed numpy-1.19.0 torch-1.4.0
[jetson-inference]  cloning torchvision...
Cloning into 'torchvision-36'...
remote: Enumerating objects: 3745, done.
remote: Total 3745 (delta 0), reused 0 (delta 0), pack-reused 3745
Receiving objects: 100% (3745/3745), 4.39 MiB | 3.37 MiB/s, done.
Resolving deltas: 100% (2436/2436), done.
[jetson-inference]  building torchvision for Python 3.6...
Traceback (most recent call last):
  File "setup.py", line 13, in <module>
    import torch
  File "/home/dennis/.local/lib/python3.6/site-packages/torch/__init__.py", line 79, in <module>
    from torch._C import *
**ImportError: libcudart.so.10.0: cannot open shared object file: No such file or directory**
Collecting pillow<7
Installing collected packages: pillow
Successfully installed pillow-6.2.2
 
[jetson-inference]  installation complete, exiting with status code 0
[jetson-inference]  to run this tool again, use the following commands:
 
    $ cd <jetson-inference>/build
    $ ./install-pytorch.sh

dusty-nv, 2020-06-23:
Hmm for some reason it doesnt seem to be detecting your JetPack version correctly.

Can you try changing this line to "1.4.0" like the line 2 lines above it?

https://github.com/dusty-nv/jetson-inference/blob/0a75e3059a93945dcd9570a5c9f3efdd9c33ce38/tools/install-pytorch.sh#L491

Then go to jetson-inference/build and re-run cmake ../

Then run ./install-pytorch.sh and try installing it again.  This time it should say "Downloading PyTorch v1.4.0" instead of 1.1.0

dusty-nv, 2020-06-23:
On second thought, that probably won't work. Instead, change the line linked to below to this:

install_pytorch_v140_python36_jp44

https://github.com/dusty-nv/jetson-inference/blob/0a75e3059a93945dcd9570a5c9f3efdd9c33ce38/tools/install-pytorch.sh#L522

https://github.com/dusty-nv/jetson-inference/blob/0a75e3059a93945dcd9570a5c9f3efdd9c33ce38/tools/install-pytorch.sh#L491

DennisFaucher, 2020-06-23:
Will do. Reflashing Xavier to Jetpack 4.4 now for a clean start

DennisFaucher, 2020-06-24:
OK. Adding Cython wasn't enough. I'll try making that L522 change above.

$ cd jetson-inference/build
$ ./install-pytorch.sh

[jetson-inference]  building torchvision for Python 3.6...
[sudo] password for dennis: 
Traceback (most recent call last):
  File "setup.py", line 13, in <module>
    import torch
  File "/home/dennis/.local/lib/python3.6/site-packages/torch/__init__.py", line 79, in <module>
    from torch._C import *
ImportError: libcudart.so.10.0: cannot open shared object file: No such file or directory

DennisFaucher, 2020-06-24:
@dusty-nv Interesting. I looked at the L522 change and realized that the install-pytorch.sh in my git clone is not the same as the install-pytorch.sh at that link. I replaced mine with your and ran as is. No run errors. 

>>> import torch
>>> print (torch. __version__)
1.4.0a0+7f73f1d

dusty-nv, 2020-06-24:
OK gotcha, glad you got it working.  That script was updated about a month ago to detect the board's JetPack version and install the appropriate version of PyTorch.

closed 2020-06-24 by DennisFaucher

flurpo, 2020-06-28:
I just updated my nano to 4.4DP bypassed the ./install-pytorch.sh and used the exact wheel file example at https://forums.developer.nvidia.com/t/pytorch-for-jetson-nano-version-1-5-0-now-available/72048. It's still pointing to libcudart.so.10.0. Just thought I'd drop that here. I'll go back to JP4.3 until this is correct......Best

dusty-nv, 2020-06-28:
If it was pointing to CUDA 10.0, then the wrong wheel was downloaded.  The wheel for JetPack 4.4 should be downloaded, and it will point to 10.2.

flurpo, 2020-06-30:
ah...that makes sense...my bad...thanks dusty-nv..