dusty-nv/jetson-inference #623: install-pytorch.sh issues with yolov5

DennisFaucher, 2020-06-24:
Feel free to send me somewhere else as this is off-topic.

I am trying to run the yolov5 examples and trianing on my Xavier from this repo: https://github.com/ultralytics/yolov5. The first big hurdle was getting pytorch installed which the jetson-interface script fixed. Now that all python libraries are installed, running "python3 detect.py --source ./inference/images/ --weights yolov5s.pt --conf 0.4" starts, gives a torch warning and core dumps. Any ideas? TIA

````[Python]
$ python3 detect.py --source ./inference/images/ --weights yolov5s.pt --conf 0.4
Namespace(agnostic_nms=False, augment=False, classes=None, conf_thres=0.4, device='', fourcc='mp4v', img_size=640, iou_thres=0.5, output='inference/output', save_txt=False, source='./inference/images/', view_img=False, weights='yolov5s.pt')
Using CUDA device0 _CudaDeviceProperties(name='Xavier', total_memory=15814MB)

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   408    0   408    0     0    440      0 --:--:-- --:--:-- --:--:--   439
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100 14.4M    0 14.4M    0     0  3167k      0 --:--:--  0:00:04 --:--:-- 5333k
Downloading https://drive.google.com/uc?export=download&id=1R5T6rIyy3lLwgFXNms8whc-387H0tMQO as yolov5s.pt... Done (8.9s)
/home/dennis/.local/lib/python3.6/site-packages/torch/serialization.py:593: SourceChangeWarning: source code of class 'torch.nn.modules.conv.Conv2d' has changed. you can retrieve the original source code by accessing the object's source attribute or set `torch.nn.Module.dump_patches = True` and use the patch tool to revert the changes.
  warnings.warn(msg, SourceChangeWarning)
Segmentation fault (core dumped)
````

dusty-nv, 2020-06-24:
SourceChangeWarning: source code of class 'torch.nn.modules.conv.Conv2d' has changed. you can retrieve the original source code by accessing the object's source attribute or set `torch.nn.Module.dump_patches = True` and use the patch tool to revert the changes.

I know you can get that error when the older-style PyTorch serialization is used where the model definition is wrapped up into the checkpoint.  Then when code changes underneath in the pytorch implementation it can lead to undefined behavior.

That's why PyTorch recommends to use load_state_dict instead - see this page for more info:   https://pytorch.org/tutorials/beginner/saving_loading_models.html

With respect to this particular model, it usually needs saved using the newer method, although I am not familiar with the yolov5 repo.

DennisFaucher, 2020-06-24:
It's alive!! I followed the instructions at https://github.com/ultralytics/yolov5/wiki/Train-Custom-Data to train my own model and the training is working! I used the same chicken data and labels that I described here: https://github.com/DennisFaucher/ChickenDetection

Youtube video of training on the Xavier can be found here: https://youtu.be/0XXQYSdQCr8

Thanks for the help.

dusty-nv, 2020-06-24:
Cool, great idea! - we have chickens too :)

DennisFaucher, 2020-06-25:
Update: Ran training for 600 iterations on the Xavier. Model works but the accuracy and performance of this detect.py on an RTSP stream is inferior to the old darknet C binary. Closing issue. Thanks for all the help.

closed 2020-06-25 by DennisFaucher