dusty-nv/jetson-inference #624: Object detection results returned as zero

curiosity-rover, 2020-06-25:
Hi,

I can see the results being displayed in the window when using DetectNet but when I try to print out the attributes like Area, centroid, etc. it is all returned as zero.

```
import jetson.inference
import jetson.utils

net=jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera=jetson.utils.gstCamera(1280, 720, "/dev/video1") 

while 1:
	img, width, height= camera.CaptureRGBA()
	detections= net.Detect(img, width, height)
	values=net.Detection()
	print(values)
        print(values.Area)
```

The output of above: 
```
   -- ClassID: 0
   -- Confidence: 0
   -- Left:    0
   -- Top:     0
   -- Right:   0
   -- Bottom:  0
   -- Width:   0
   -- Height:  0
   -- Area:    0
   -- Center:  (0, 0)
0.0
```

curiosity-rover, 2020-06-25:
ok,so
`print(detections[0])`
gives me the results, so I'm guessing the net.Detection method that I used is not the correct one for reading?
But, the net.Detect method should return a list and, in this case, it seems like the list doesn't have 0 items ...

curiosity-rover, 2020-06-25:
ok, I realised what I did wrong. I was only detecting 1 object so I should have done:
`print(detections[0].Area)`
which works now

closed 2020-06-25 by curiosity-rover

shsur-27, 2020-07-17:
hi, i am getting the same problem as when i used
  `detections=` net.Detect(img, width, height)
	values=net.Detection()
	print(values)
        print(values.Area)`

i get 
`<detectNet.Detection object>
   -- ClassID: 0
   -- Confidence: 0
   -- Left:    0
   -- Top:     0
   -- Right:   0
   -- Bottom:  0
   -- Width:   0
   -- Height:  0
   -- Area:    0
   -- Center:  (0, 0)`

then i tried
`print(detections[0])`

and i got:
`  File "video-detection.py", line 26, in <module>
    print(detections[0])
IndexError: list index out of range`

same error for when i tried 
`print(detections[0].Area)`

`  File "video-detection.py", line 26, in <module>
    print(detections[0].Area)
IndexError: list index out of range`

curiosity-rover, 2020-07-17:
Your last two print statements should work. 'Index out of range' most likely means that the list 'detections' is empty (no objects detected).

Try using a 'try...except' like

`try:
    print(detections[0])
except IndexError as e:
    print(e)`

dusty-nv, 2020-07-17:
Yes try print(len(detections)) also