dusty-nv/jetson-inference #625: Using detectNet-camera with ZED2 Stereoscopic Camera (python)

birdistheword96, 2020-06-25:
I have seen a few posts on here about using the ZED cameras but none have been very clear on the outcome, and what/if anything needs modifying in the source code to get it working. Does anyone have a basic guide on how to get started? 

ZED have a library (wrapper) which allows you to grab images from the camera and it works really well, but I think they must be in the wrong format for `GLDisplay().` How do I get it so I can least see a live video output?

The ZED library has a useful` 'get_data()' `function which converts the image into a numpy array [Link to ZED Docs](https://www.stereolabs.com/docs/opencv/python/#sharing-image-data-between-zed-sdk-and-opencv-python), and it works for open CV, so I was working under the assumption that i could use `from jetson.utils import cudaFromNumpy` to convert the image into a recognised format for `GDisplay()`?

Here's what I have so far, which is a modified version of the detectNet-Camera example file:

``` python
import jetson.inference
import jetson.utils
from jetson.utils import cudaFromNumpy as cfn, saveImageRGBA
import pyzed.sl as sl

import argparse
import sys

parser = argparse.ArgumentParser(description="Locate objects in a live camera stream using an object detection DNN.", 
						   formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.detectNet.Usage())

parser.add_argument("--network", type=str, default="ssd-mobilenet-v2", help="pre-trained model to load (see below for options)")
parser.add_argument("--overlay", type=str, default="box,labels,conf", help="detection overlay flags (e.g. --overlay=box,labels,conf)\nvalid combinations are:  'box', 'labels', 'conf', 'none'")
parser.add_argument("--threshold", type=float, default=0.5, help="minimum detection threshold to use") 
parser.add_argument("--camera", type=str, default="0", help="index of the MIPI CSI camera to use (e.g. CSI camera 0)\nor for VL42 cameras, the /dev/video device to use.\nby default, MIPI CSI camera 0 will be used.")
parser.add_argument("--width", type=int, default=1280, help="desired width of camera stream (default is 1280 pixels)")
parser.add_argument("--height", type=int, default=720, help="desired height of camera stream (default is 720 pixels)")

try:
	opt = parser.parse_known_args()[0]
except:
	print("")
	parser.print_help()
	sys.exit(0)

# load the recognition network
net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)

zed = sl.Camera()
init_params = sl.InitParameters()
init_params.camera_resolution = sl.RESOLUTION.HD1080  # Use HD1080 video mode
init_params.camera_fps = 30  # Set fps at 30
zed.open()

# Create a matrix to store the image
image_zed = sl.Mat()
width = opt.width
height = opt.height
# create the camera and display
font = jetson.utils.cudaFont()
#camera = jetson.utils.gstCamera(opt.width, opt.height, opt.camera)
display = jetson.utils.glDisplay()

# process frames until user exits
while display.IsOpen():
	# capture the image
	zed.grab()
	zed.retrieve_image(image_zed, sl.VIEW.LEFT)
	image_data = image_zed.get_data()
	cuda_mem = cfn(image_data)

	detections = net.Detect(cuda_mem, width, height, opt.overlay)

	# print the detections
	print("detected {:d} objects in image".format(len(detections)))

	for detection in detections:
		print(detection)

	# render the image
	display.RenderOnce(image, width, height)

	# update the title bar
	display.SetTitle("{:s} | Network {:.0f} FPS".format(opt.network, net.GetNetworkFPS()))

	# print out performance info
	net.PrintProfilerTimes()
```

Unfortunately this causes all kinds of errors and never opens the display, errors ranging from

`Exception: jetson.utils -- cudaAllocMapped() failed to parse size argument
PyTensorNet_Dealloc()`

to 

`Exception: jetson.inference -- detectNet.Detect() encountered an error classifying the image`

I apologise if this is a bit of a noob question and is missing any information, I just don't know where to go from here or what the issue actually is, so grateful fro any help!

Has anybody had any experience in using the ZED Cameras and managed to get them work with Jetson Inference?

dusty-nv, 2020-06-25:
I think you need a call to `cv2.cvtColor(image_data, cv2.COLOR_BGR2RGBA)` before you call `cudaFromNumpy()`.

If you still get errors after doing that, please post the log in it's entirety.  Thanks.

birdistheword96, 2020-06-25:
Thanks for the reply, unfortunately it didn't work: 
```
jetson.inference.__init__.py
jetson.inference -- initializing Python 3.6 bindings...
jetson.inference -- registering module types...
jetson.inference -- done registering module types
jetson.inference -- done Python 3.6 binding initialization
jetson.utils.__init__.py
jetson.utils -- initializing Python 3.6 bindings...
jetson.utils -- registering module functions...
jetson.utils -- done registering module functions
jetson.utils -- registering module types...
jetson.utils -- done registering module types
jetson.utils -- done Python 3.6 binding initialization
jetson.inference -- PyTensorNet_New()
jetson.inference -- PyDetectNet_Init()
jetson.inference -- detectNet loading network using argv command line params
jetson.inference -- detectNet.__init__() argv[0] = 'jetson_test.py'

detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

[TRT]   TensorRT version 6.0.1
[TRT]   loading NVIDIA plugins...
[TRT]   Plugin Creator registration succeeded - GridAnchor_TRT
[TRT]   Plugin Creator registration succeeded - GridAnchorRect_TRT
[TRT]   Plugin Creator registration succeeded - NMS_TRT
[TRT]   Plugin Creator registration succeeded - Reorg_TRT
[TRT]   Plugin Creator registration succeeded - Region_TRT
[TRT]   Plugin Creator registration succeeded - Clip_TRT
[TRT]   Plugin Creator registration succeeded - LReLU_TRT
[TRT]   Plugin Creator registration succeeded - PriorBox_TRT
[TRT]   Plugin Creator registration succeeded - Normalize_TRT
[TRT]   Plugin Creator registration succeeded - RPROI_TRT
[TRT]   Plugin Creator registration succeeded - BatchedNMS_TRT
[TRT]   Could not register plugin creator:  FlattenConcat_TRT in namespace: 
[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - UFF  (extension '.uff')
[TRT]   desired precision specified for GPU: FASTEST
[TRT]   requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]   native precisions detected for GPU:  FP32, FP16, INT8
[TRT]   selecting fastest native precision for GPU:  FP16
[TRT]   attempting to open engine cache file /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.GPU.FP16.engine
[TRT]   loading network profile from engine cache... /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.GPU.FP16.engine
[TRT]   device GPU, /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff loaded
[TRT]   Deserialize required 2292775 microseconds.
[TRT]   device GPU, CUDA engine context initialized with 3 bindings
[TRT]   binding -- index   0
               -- name    'Input'
               -- type    FP32
               -- in/out  INPUT
               -- # dims  3
               -- dim #0  3 (SPATIAL)
               -- dim #1  300 (SPATIAL)
               -- dim #2  300 (SPATIAL)
[TRT]   binding -- index   1
               -- name    'NMS'
               -- type    FP32
               -- in/out  OUTPUT
               -- # dims  3
               -- dim #0  1 (SPATIAL)
               -- dim #1  100 (SPATIAL)
               -- dim #2  7 (SPATIAL)
[TRT]   binding -- index   2
[... 34 lines cut ...]
[cuda]     invalid resource handle (error 33) (hex 0x21)
[cuda]     /home/user/jetson-inference/c/detectNet.cpp:637
[TRT]   detectNet::Detect() -- cudaPreImageNetNorm() failed
Traceback (most recent call last):
  File "jetson_test.py", line 57, in <module>
    detections = net.Detect(cuda_mem, width, height, opt.overlay)
Exception: jetson.inference -- detectNet.Detect() encountered an error classifying the image
PyTensorNet_Dealloc()
jetson.utils -- PyFont_Dealloc()
[cuda]  cudaFreeHost(mRectsCPU)
[cuda]     invalid argument (error 11) (hex 0x0B)
[cuda]     /home/user/jetson-inference/utils/cuda/cudaFont.cu:96
[cuda]  cudaFreeHost(mCommandCPU)
[cuda]     invalid argument (error 11) (hex 0x0B)
[cuda]     /home/user/jetson-inference/utils/cuda/cudaFont.cu:104
[cuda]  cudaFreeHost(mFontMapCPU)
[cuda]     invalid argument (error 11) (hex 0x0B)
[cuda]     /home/user/jetson-inference/utils/cuda/cudaFont.cu:112
jetson.utils -- PyDisplay_Dealloc()
jetson.utils -- freeing CUDA mapped memory
[cuda]  cudaFreeHost(ptr)
[cuda]     invalid argument (error 11) (hex 0x0B)
[cuda]     /home/user/jetson-inference/utils/python/bindings/PyCUDA.cpp:121
jetson.utils -- failed to free CUDA mapped memory with cudaFreeHost()
```

dusty-nv, 2020-06-25:
OK, let's see first if you can just display the image with glDisplay first.  Comment out the detectNet stuff for now:

```python
	#detections = net.Detect(cuda_mem, width, height, opt.overlay)

	# print the detections
	#print("detected {:d} objects in image".format(len(detections)))

	#for detection in detections:
	#	print(detection)

	# render the image
	display.RenderOnce(cuda_mem, width, height)

	# update the title bar
	#display.SetTitle("{:s} | Network {:.0f} FPS".format(opt.network, net.GetNetworkFPS()))

	# print out performance info
	#net.PrintProfilerTimes()
```

birdistheword96, 2020-06-25:
Yes, that works! 

I get a lot of:
```
jetson.utils -- freeing CUDA mapped memory
jetson.utils -- cudaFromNumpy()  ndarray dim 0 = 720
jetson.utils -- cudaFromNumpy()  ndarray dim 1 = 1280
jetson.utils -- cudaFromNumpy()  ndarray dim 2 = 4
```
printed to the screen but I can actually see the image, which is more than before! So I assume the next step is to work out why the net.Detect() function doesn't like the image but the display does?

dusty-nv, 2020-06-25:
> So I assume the next step is to work out why the net.Detect() function doesn't like the image but the display does?

Yep correct, I am at a loss for why glDisplay would work but detectNet doesn't.   My only thought is perhaps ZED SDK is initializing the CUDA context in a way that is conflicting.

Can you try moving the `net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)` line below `zed.open()`?

birdistheword96, 2020-06-26:
Surprisingly that actually worked, moving it below allowed the image to appear and a bounding box drawn on screen for an object.

The final issue (not sure if its related to the ZED camera so happy to move to separate issue if required) is that I am unable to exit the program properley, if I try to close the window (or press ctrl-c) at the command line then the whole program freezes, even the command line won't force it to quit, and then some time later python crashes. Its not until about 2 or 3 mins later that the frozen gldisplay window closes itself. This is the error message printed out when I try to quit:

```
[OpenGL] glDisplay -- the window has been closed
PyTensorNet_Dealloc()
[cuda]  cudaFreeHost(mDetectionSets[0])
[cuda]     invalid argument (error 11) (hex 0x0B)
[cuda]     /home/racelogic/jetson-inference/c/detectNet.cpp:66
[cuda]  cudaFreeHost(mClassColors[0])
[cuda]     invalid argument (error 11) (hex 0x0B)
[cuda]     /home/racelogic/jetson-inference/c/detectNet.cpp:74
^CSegmentation fault (core dumped)
```

Is this something you've seen before? Do I need to call a cleanup function before quitting?

birdistheword96, 2020-06-26:
For clarity I'm running **Ubuntu 18.04.4 LTS** with python 3.6

dusty-nv, 2020-06-26:
> Is this something you've seen before? Do I need to call a cleanup function before quitting?

Unfortunately I haven't seen that before, and you don't need to call a cleanup function yourself.   These errors though are similar to the ones you were getting before (invalid cuda resource), and it seems ZED SDK and jetson-inference have some kind of interference going on somewhere.

What Jetson and version of JetPack are you running btw?

birdistheword96, 2020-06-30:
I'm running Jetpack 4.3 on an Nvidea Jetson Xavier

canozcivelek, 2020-07-07:
Hi, I have a similar problem too. I'm on a Jetson Nano, the inference examples work fine, segnet-console.py example also works without problem. However, running "python segnet-camera.py --network=fcn-resnet18-cityscapes" command gives the following error:

Traceback (most recent call last):
  File "segnet-camera.py", line 58, in <module>
    img_mask = jetson.utils.cudaAllocMapped(opt.width/2 * opt.height/2 * 4 * ctypes.sizeof(ctypes.c_float))
Exception: jetson.utils -- cudaAllocMapped() failed to parse size argument
jetson.inference -- PySegNet_Dealloc()
jetson.utils -- freeing CUDA mapped memory

This isn't the whole log though, I can share it if needed. I have a Logitech webcam attached which works fine with other examples. Can you help me out? Thanks a lot!

dusty-nv, 2020-07-07:
> File "segnet-camera.py", line 58, in
> img_mask = jetson.utils.cudaAllocMapped(opt.width/2 * opt.height/2 * 4 * ctypes.sizeof(ctypes.c_float))

Hi @canozcivelek, this issue had been fixed in commit [`513707`](https://github.com/dusty-nv/jetson-inference/commit/513707685db6d4e009539c504b09851ce8dacfbb).  Can you try pulling the latest from master?

canozcivelek, 2020-07-08:
Hi again @dusty-nv, thanks for the quick response! I pulled the latest version and the error is gone. However, now it's giving me "Exception: jetson.inference -- segNet failed to load network" error. I checked the /data/networks/ folder and the network I'm providing is --network=fcn-resnet18-cityscapes which exists in the folder. I am running the script from /python/examples dir. Is it because of some sort of  path related problem or am I missing something here? Thanks again!

closed 2023-04-27 by dusty-nv