dusty-nv/jetson-inference #626: Use newest jetson-utils in jetson-inference

Philliec459, 2020-06-28:
I am wondering if I should be updating jetson-inference, especially using the new code in jetson-utils?  I have read that there are a few issues that have been fixed using the newest jetson-utils such as displaying a small video window and not having the window trim take up the whole display. 

To update my jetson-inference I could try to copy jetson-utils into a new download of jetson-inference, and then build jetson-inference from source. Before trying this I thought that I should seek your advice.

dusty-nv, 2020-06-28:
Hi, use the dev branch of jetson-inference if you want the latest jetson-utils.  Thanks.

Philliec459, 2020-06-28:
Great. Thanks.

Philliec459, 2020-06-29:
worked perfect. Thanks.