dusty-nv/jetson-inference #635: Filtering ssd-mobilenet-v2

danishsayed, 2020-07-06:
How would I go about detecting only a few of the [91 classes ](https://github.com/dusty-nv/jetson-inference/blob/master/data/networks/ssd_coco_labels.txt) available for the ssd-mobilenet-v2? 

I am currently using the code available in the tutorial portion: 

`import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.gstCamera(1280, 720, "0")
display = jetson.utils.glDisplay()

while display.IsOpen():
	img, width, height = camera.CaptureRGBA()
	detections = net.Detect(img, width, height)
	display.RenderOnce(img, width, height)
	display.SetTitle("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
`

dusty-nv, 2020-07-06:
The best way would be to re-train the model on only classes that you are interested in - I am currently working on the re-training model.

But in lieu of that, go into the labels file for the model and change the names of the classes that you want to ignore to `void`.

The labels file for ssd-mobilenet-v2 is found under `jetson-inference/data/networks/SSD-Mobilenet-v2/ssd_coco_labels.txt`

danishsayed, 2020-07-07:
Thank you so much @dusty-nv - awesome tutorials btw, this stuff is amazing!

Abud17, 2021-11-04:
> The best way would be to re-train the model on only classes that you are interested in - I am currently working on the re-training tutorial for object detection on SSD-Mobilenet using PyTorch.
> 
> But in lieu of that, go into the labels file for the model and change the names of the classes that you want to ignore to `void`.
> 
> The labels file for ssd-mobilenet-v2 is found under `jetson-inference/data/networks/SSD-Mobilenet-v2/ssd_coco_labels.txt`

hi Dusty-nv 
is there a vedio of yours for re-training SSD-Mobilenet with a specific class?

dusty-nv, 2021-11-05:
Hi @Abud17, here is a video and tutorial about re-training SSD-Mobilenet:

* https://www.youtube.com/watch?v=2XMkPW_sIGg&list=PL5B692fm6--uQRRDTPsJDp4o0xbzkoyf8&index=13
* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md

Abud17, 2021-11-06:
Thanks a lot sir 😊

https://www.youtube.com/watch?v=2XMkPW_sIGg&list=PL5B692fm6--uQRRDTPsJDp4o0xbzkoyf8&index=13https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md

closed 2023-04-27 by dusty-nv