dusty-nv/jetson-inference #636: jetson xaiver build error GL/glew.h: No such file or directory

hushunda, 2020-07-07:
[  1%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glDisplay.cpp.o
In file included from /home/nvidia/jetson-inference/utils/display/glDisplay.h:27:0,
                 from /home/nvidia/jetson-inference/utils/display/glDisplay.cpp:23:
/home/nvidia/jetson-inference/utils/display/glUtility.h:27:10: fatal error: GL/glew.h: No such file or directory
 #include <GL/glew.h>
          ^~~~~~~~~~~
compilation terminated.
utils/CMakeFiles/jetson-utils.dir/build.make:2241: recipe for target 'utils/CMakeFiles/jetson-utils.dir/display/glDisplay.cpp.o' failed
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/display/glDisplay.cpp.o] Error 1
CMakeFiles/Makefile2:854: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

who can help me?

dusty-nv, 2020-07-07:
Can you try running `sudo apt-get install libglew-dev`?

This should have been installed automatically when you ran `cmake ../`, so I'm not sure why it isn't.

hushunda, 2020-07-08:
thank your reply!I run sudo apt-get install libglew-dev and Its  ok.
but I have  other build problem.

`
/usr/local/include/NvOnnxParser.h:27:10: fatal error: NvOnnxParserTypedefs.h: No such file or directory
 #include "NvOnnxParserTypedefs.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~
`
i need to install ONNX?Or only some package like libglew-dev?

dusty-nv, 2020-07-08:
This file is part of TensorRT, which should have been installed automatically when you flashed JetPack.  It seems like not the entire JetPack was flashed to your board.

closed 2020-07-27 by hushunda

3r0s3Nn1n, 2021-08-23:
I had the same error i was trying to use both docker and source when i got the message. I removed the files and followed just building the project from source steps and it works now!