dusty-nv/jetson-inference #638: Auto-resizing of object detection window

dummy-jetbot, 2020-07-10:
I am using a USB camera and it supports only 640 x 480, which looks very small on the monitor.
May I know if there any way to re-size the frame to window size, .e.g auto-resizing to glDisplay to fit?

dusty-nv, 2020-07-12:
Hi @dummy-jetbot, if you clone and install the `dev` branch of jetson-inference, it will do the auto-resizing.

dummy-jetbot, 2020-07-13:
@dusty-nv Thanks so much for your reply! 
Sorry but may I know how to clone and install the `dev` branch of jetson-inference?

dusty-nv, 2020-07-13:
git clone --branch dev https://github.com/dusty-nv/jetson-inference jetson-inference-dev

Then build and install it like normal (it will be in jetson-inference-dev folder)

closed 2023-04-27 by dusty-nv