dusty-nv/jetson-inference #641: Segmentation fault when running detectnet with a caffemodel

BullotLo, 2020-07-16:
Hi,
I am trying to run the detection example on my jetson nano with detectnet.py and a model I trained in DIGITS with the, now deprecated, two day to a demo tutorial. It worked fine before the last update of 15.07.2020 but now when I try to run this command:

**$ python3 detectnet.py --prototxt=$NET/deploy.prototxt --model=$NET/snapshot_iter_52000.caffemodel**

The program start and after getting the first frame it crash with a segmentation fault as below.

[TRT]    ------------------------------------------------
[TRT]    Timing Report /home/jetson/jetson-inference/build/aarch64/bin/networks/Mouse_1000ep_nosub/snapshot_iter_52000.caffemodel
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.05917ms  CUDA   6.55312ms
[TRT]    Network       CPU 150.64021ms  CUDA 136.23354ms
[TRT]    Post-Process  CPU   0.28928ms  CUDA   0.28932ms
[TRT]    Total         CPU 150.98866ms  CUDA 143.07599ms
[TRT]    ------------------------------------------------

[TRT]    note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

Segmentation fault (core dumped)

After that even trying to run the detection with the default model don't work anymore I have to restart the jetson nano.
Any idea where it could come from ?

dusty-nv, 2020-07-16:
Hmm, does it work if you just process a single image?  Like this:

```
$ python3 detectnet.py --prototxt=$NET/deploy.prototxt --model=$NET/snapshot_iter_52000.caffemodel my_image.jpg
```

Or what happens if you run the C++ version (`./detectnet`) instead of Python version?

If you continue to have the issue, for now you can revert to using the `20200714` branch (which I backed-up before I merged all these changed):   https://github.com/dusty-nv/jetson-inference/tree/20200714

BullotLo, 2020-07-17:
I tried with a few different images from the image folder and also with some of mine. It appears that whenever the model doesn't detect an object in the image he runs fine (the model is trained to detect only 1 thing, a laboratory mouse) but when I use my images or an image where he could detect an object, I get a segmentation fault.

When it work:
[image] loaded 'images/airplane_1.jpg'  (427x640, 3 channels)
detected 0 objects in image
[OpenGL] glDisplay -- set the window size to 427x640
[OpenGL] creating 427x640 texture (GL_RGB8 format, 819840 bytes)
[cuda]   registered openGL texture for interop access (427x640, GL_RGB8, 819840 bytes)
[image] saved 'output_0.jpg'  (427x640, 3 channels)

[TRT]    ------------------------------------------------
[TRT]    Timing Report /home/jetson/jetson-inference/build/aarch64/bin/networks/Mouse_1000ep_nosub/snapshot_iter_52000.caffemodel
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.08297ms  CUDA   9.29062ms
[TRT]    Network       CPU 198.47118ms  CUDA 189.13553ms
[TRT]    Post-Process  CPU   0.25839ms  CUDA   0.25771ms
[TRT]    Total         CPU 198.81255ms  CUDA 198.68385ms
[TRT]    ------------------------------------------------

[TRT]    note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

When it doesn't:
[image] loaded 'images/cat_1.jpg'  (350x441, 3 channels)
Segmentation fault (core dumped)

The c++ version ./detectnet behave exactly the same
I'v also tried different models I trained and the result is the same

What surprise me the most is that I get the same problem when I use detectnet-console with the 20200714 branch. I don't understand why. I initially pulled the jetson-inference master on June 18. I'm not extremely familiar with github but is it possible for me to get the version at this point of time ?

dusty-nv, 2020-07-17:
The submodules are automatically reverted to their older versions when you `git checkout 0a75e3059a93945dcd9570a5c9f3efdd9c33ce38`

Just remember to re-compile and do `sudo make install` so you don't have another version of the shared libraries installed on your system.

BullotLo, 2020-07-17:
Thanks,

I could get back to a point where my inference with a custom model work. I'm gonna work with that from now on.

That doesn't resolve the problem tho, do you want me to close the issue anyway ?

s183724, 2021-09-22:
Hi @dusty-nv - i seem to be having the same problem, as described above. 

1. I know that it is not a memory issue (ive been running it with tegrastats, and swap file still has plenty of space)
2. I know that it is not a corrupt model. Ive been using the model that you trained in https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md#training-the-ssd-mobilenet-model
3. I am succesfully able to run the program with the model, on images that do not contain the classes. However the program crashes when run on images containing the classes with the error: "Segmentation fault (core dumped)"

I am running on an nvidia jetson nano 2GB, and i have been following the youtube tutorial (hello world AI) pretty strictly. Is the code still working properly on your setup?

dusty-nv, 2021-09-22:
Hi @s183724, what is the command line that you are running?

s183724, 2021-09-22:
Hi @dusty-nv. Thank you so much for the quick reply! 

I have tried multiple variations of the command, both with my own images downloaded into the data folder and the preinstalled images. I have also tried the specific command you use in your youtube video (which has been a great help, THANKS!). 

i am running this command from within the docker container: 
detectnet --model=models/fruit/ssd-mobilenet.onnx --labels=models/fruit/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes "/jetson-inference/data/images/fruit_*.jpg" /jetson-inference/data/images/test/fruit_%i.jpg

If i feed this command pictures of non-fruits it succesfully goes to the next image, but when i feed it a fruit i get:
[image] loaded 'images/fruit_1.jpg' (350x441, 3 channels)
Segmentation fault (core dumped)

closed 2023-04-27 by dusty-nv