dusty-nv/jetson-inference #643: Camera Feed GUI Doesn't Open

jasonotu, 2020-07-19:
Hi,

I've re-trained the SSD model to use a small subset of the classes, however, when I run detectnet or detectnet-camera, it is able to detect from the feed but the GUI live feed window never opens. It opens fine using other demos or networks.

`detectnet-camera.py --model=models/test/ssd-mobilenet.onnx --labels=models/test/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes --input-width=1920 --input-height=1080 csi://0`

Sample output excerpt:

`1 objects detected
detected obj 0  class #1 (Class 1)  confidence=0.621149
bounding box 0  (96.445312, 117.861328)  (420.919312, 827.796570)  w=324.473999  h=709.935242

[TRT]    ------------------------------------------------
[TRT]    Timing Report models/test/ssd-mobilenet.onnx
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.11740ms  CUDA   1.21818ms
[TRT]    Network       CPU  36.75002ms  CUDA  25.89125ms
[TRT]    Post-Process  CPU   1.37012ms  CUDA   1.41943ms
[TRT]    Visualize     CPU   0.18995ms  CUDA   3.29745ms
[TRT]    Total         CPU  38.42748ms  CUDA  31.82630ms
[TRT]    ------------------------------------------------
`

Any ideas?

jasonotu, 2020-07-22:
I was able to get around this by adding a video output name to the end:

`detectnet-camera.py --model=models/test/ssd-mobilenet.onnx --labels=models/test/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes --input-width=1920 --input-height=1080 csi://0 **output.mp4**`

Wasn't sure if I can get around creating the video so I just used /dev/null.

closed 2020-07-22 by jasonotu

dusty-nv, 2020-07-22:
Thanks, I'm curious - does it also make the GUI if you launch it with `display://0` at the end (instead of `output.mp4`)

jasonotu, 2020-07-24:
Yes, it does.

dusty-nv, 2020-07-24:
OK, thanks.   I made a fix for this yesterday, so if you `git pull origin master` and then `make` and `sudo make install`, it should be working without needing the `display://0` anymore.

jasonotu, 2020-07-24:
Awesome, thanks!