dusty-nv/jetson-inference #644: Example of creating a custom object detector from scratch

manrayboy, 2020-07-19:
I'm having issues creating an object detector from scratch. I used labelimg to annotate my images and then tried to create my own xml_to_csv to convert the columns similar to the: sub-test-annotations-bbox.csv and sub-train-annotations-bbox.csv.

I'm really close but running into issues like: AttributeError: Can't pickle local object 'TrainAugmentation.__init__.<locals>.<lambda>'
and so on. We're kind of shooting in the dark here. If you have any example of creating this from scratch that would be awesome

dusty-nv, 2020-07-19:
@manarayboy see this updated documentation - https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md

manrayboy, 2020-07-19:
Yeah, i've read that actually, but that's only for camera-capture. I already have all the images that I downloaded from google:
In that tutorial it says this though, but labelimg doesn't convert in a format even close to the one pytorch-ssd needs

`note: if you want to label a set of images that you already have (as opposed to capturing them from camera), try using a tool like LabelImg that also saves in Pascal VOC format. If you need to label a video file, dump the video frames to images first.`

dusty-nv, 2020-07-19:
LabelImg supports Pascal VOC format.  pytorch-ssd supports Pascal VOC (--dataset-type=voc) and Open Images

Philliec459, 2020-07-21:
@manrayboy, I will create a GitHub repository with my marine implementation of pytorch-ssd. Maybe it was dumb luck, but it seems to be working just fine. It appears a bit slow so I am very interested in applying @dusty-nv 's new Re-training SSD-Mobilenet that was mentioned late last week once I figure out how to load JP4.4 with my SSD.

Philliec459, 2020-07-22:
@manrayboy, I have searched through my notes and have the beginning of a GitHub repository where we have taken @dusty-nv pytorch-ssd and retrained using marine objects (boats and buoys). Please add @dusty-nv vision and models subdirectories. I will upload the marine training/test dataset and models to AWS if needed. Please go to:

https://github.com/Philliec459/Pytorch-SSD-for-Marine-Objects/blob/master/README.md

closed 2023-04-27 by dusty-nv