dusty-nv/jetson-inference #645: videoSource not avaliable with latest jetson-inference

mmatesic01, 2020-07-19:
Greetings,

I have installed jetson-inference on 18th July 2020 by these instructions.
https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md

But when I use videoSource in my python (for example cam = jetson.utils.videoSource('RTSP LINK') I get error message that videoSource is "unknown" to  jetson.utils.
I have cheked my downloaded build and it is the same as this one: https://github.com/dusty-nv/jetson-inference

What am I doing wrong here?

Thanks.

dusty-nv, 2020-07-19:
Hmm are you able to run the other samples like video-viewer.py ?

Did you run 'sudo make install'?

https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md

mmatesic01, 2020-07-19:
Hi, 
I did the "sudo make install" at the end.

With video.viewer.py I get this: 
jetson.utils.__init__.py
jetson.utils -- initializing Python 3.6 bindings...
jetson.utils -- registering module functions...
jetson.utils -- done registering module functions
jetson.utils -- registering module types...
jetson.utils -- done registering module types
jetson.utils -- done Python 3.6 binding initialization
Traceback (most recent call last):
  File "video-viewer.py", line 33, in <module>
    epilog=jetson.utils.videoSource.Usage() + jetson.utils.videoOutput.Usage() + jetson.utils.logUsage())
AttributeError: module 'jetson.utils' has no attribute 'videoSource'

mmatesic01, 2020-07-20:
HI,
I reinstalled from scratch and now it is working, to one extent.
It is an RTSP stream from Reolink camera with h264 codec.
Now I get this error.

[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for xxx:xxx@192.168.2.xxx
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
[gstreamer] gstDecoder -- failed to discover stream info
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://xxxxx:xxxx@192.168.2.xxx:554/h264Preview_01_main
Traceback (most recent call last):
  File "/home/marko/Desktop/pyProg/NVIDIA/deepLearning-4.py", line 13, in <module>
    cam = jetson.utils.videoSource('rtsp://xxxxx:xxxx@192.168.2.xxx:554/h264Preview_01_main')
Exception: jetson.utils -- failed to create videoSource device

dusty-nv, 2020-07-20:
It seems unable to connect to your camera to discover the stream properties.

What happens if you run it with the `--input-codec=h264` also?

mmatesic01, 2020-07-20:
Hi Dusty,
How do I input this in to python code:
cam = jetson.utils.videoSource('rtsp://xxx:xxxx@192.168.2.xxx:554/h264Preview_01_main')

dusty-nv, 2020-07-20:
See how video-viewer passes `sys.argv` to `videoSource` here:

https://github.com/dusty-nv/jetson-utils/blob/a587c20ad95d71efd47f9c91e3fbf703ad48644d/python/examples/video-viewer.py#L47

If you want to hard-code it, you may be able to do something like this:

`cam = jetson.utils.videoSource('rtsp://xxx:xxxx@192.168.2.xxx:554/h264Preview_01_main', argv=['--input-codec=h264'])`

mmatesic01, 2020-07-20:
I was going in the right direction with "argv", but just did not know how to format it right.
Now it works as expected.
Thank you.

mmatesic01, 2020-07-20:
Never mind I solved it myself. Thank you.

closed 2020-07-20 by mmatesic01