dusty-nv/jetson-inference #649: Retraining SSD-Mobilenet

arjuntx2, 2020-07-21:
Hi,I just want to make something clear:

```
detectnet.py --model=models/fruit/ssd-mobilenet.onnx --labels=models/fruit/labels.txt \
          --input-blob=input_0 --output-cvg=scores --output-bbox=boxes \
            csi://0
```

In this argument, is **detectnet.py** same as **detectnet-camera.py** when I want to run from the live camera. If so, I am getting following error.

```
jetson.inference.__init__.py
jetson.inference -- initializing Python 2.7 bindings...
jetson.inference -- registering module types...
jetson.inference -- done registering module types
jetson.inference -- done Python 2.7 binding initialization
jetson.utils.__init__.py
jetson.utils -- initializing Python 2.7 bindings...
jetson.utils -- registering module functions...
jetson.utils -- done registering module functions
jetson.utils -- registering module types...
jetson.utils -- done registering module types
jetson.utils -- done Python 2.7 binding initialization
jetson.inference -- PyTensorNet_New()
jetson.inference -- PyDetectNet_Init()
jetson.inference -- detectNet loading network using argv command line params
jetson.inference -- detectNet.__init__() argv[0] = '/usr/local/bin/detectnet-camera.py'
jetson.inference -- detectNet.__init__() argv[1] = '--network=models/fruit/ssd-mobilenet.onnx'
jetson.inference -- detectNet.__init__() argv[2] = '--labels=models/fruit/labels.txt'
jetson.inference -- detectNet.__init__() argv[3] = '--input-blob=input_0'
jetson.inference -- detectNet.__init__() argv[4] = '--output-cvg=scores'
jetson.inference -- detectNet.__init__() argv[5] = '--output-bbox=boxes'

detectNet -- loading detection network model from:
          -- prototxt     NULL
          -- model        models/fruit/ssd-mobilenet.onnx
          -- input_blob   'data'
          -- output_cvg   'coverage'
          -- output_bbox  'bboxes'
          -- mean_pixel   0.000000
          -- mean_binary  NULL
          -- class_labels NULL
          -- threshold    0.500000
          -- batch_size   1

[TRT]   TensorRT version 7.1.0
[TRT]   loading NVIDIA plugins...
[TRT]   Plugin creator registration succeeded - ::GridAnchor_TRT
[TRT]   Plugin creator registration succeeded - ::NMS_TRT
[TRT]   Plugin creator registration succeeded - ::Reorg_TRT
[TRT]   Plugin creator registration succeeded - ::Region_TRT
[TRT]   Plugin creator registration succeeded - ::Clip_TRT
[TRT]   Plugin creator registration succeeded - ::LReLU_TRT
[TRT]   Plugin creator registration succeeded - ::PriorBox_TRT
[TRT]   Plugin creator registration succeeded - ::Normalize_TRT
[TRT]   Plugin creator registration succeeded - ::RPROI_TRT
[TRT]   Plugin creator registration succeeded - ::BatchedNMS_TRT
[TRT]   Could not register plugin creator:  ::FlattenConcat_TRT
[TRT]   Plugin creator registration succeeded - ::CropAndResize
[TRT]   Plugin creator registration succeeded - ::DetectionLayer_TRT
[TRT]   Plugin creator registration succeeded - ::Proposal
[TRT]   Plugin creator registration succeeded - ::ProposalLayer_TRT
[TRT]   Plugin creator registration succeeded - ::PyramidROIAlign_TRT
[TRT]   Plugin creator registration succeeded - ::ResizeNearest_TRT
[TRT]   Plugin creator registration succeeded - ::Split
[TRT]   Plugin creator registration succeeded - ::SpecialSlice_TRT
[TRT]   Plugin creator registration succeeded - ::InstanceNormalization_TRT
[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - ONNX  (extension '.onnx')
[TRT]   desired precision specified for GPU: FASTEST
[... 97 lines cut ...]
[gstreamer] gstCamera onPreroll
[gstreamer] gstCamera -- allocated 16 ringbuffers, 1382400 bytes each
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink
[gstreamer] gstreamer msg async-done ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> mysink
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline0
[gstreamer] gstCamera -- allocated 16 RGBA ringbuffers
[TRT]   ../rtSafe/cuda/genericReformat.cu (1294) - Cuda Error in executeMemcpy: 1 (invalid argument)
[TRT]   FAILED_EXECUTION: std::exception
[TRT]   detectNet::Detect() -- failed to execute TensorRT context
Traceback (most recent call last):
  File "/usr/local/bin/detectnet-camera.py", line 61, in <module>
    detections = net.Detect(img, width, height, opt.overlay)
Exception: jetson.inference -- detectNet.Detect() encountered an error classifying the image
jetson.utils -- PyDisplay_Dealloc()
jetson.utils -- PyCamera_Dealloc()
[gstreamer] closing gstCamera for streaming, transitioning pipeline to GST_STATE_NULL
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success
PyTensorNet_Dealloc()

```

Thank you for the hepl :)

dusty-nv, 2020-07-21:
When you type that command into the console, are you running it with the `\` forward slashes at the end of the lines?

If so, remove those and just run it on one line.

dusty-nv, 2020-07-21:
Also, does it work when you run it as `detectnet.py`, but not `detectnet-camera.py`?

arjuntx2, 2020-07-21:
Yes, I am running with \ forward slashes. I will remove it and give it a try :)

Although there is no file named detectnet in Pytorch-ssd directory when I cloned it. Do I need to copy file detectnet-camera from build/aarch64/bin ? Because I am seeing bash error : No command found named detectnet!

dusty-nv, 2020-07-21:
You need to do `sudo make install`.   Maybe it is running the old detectnet-camera that hadn't been updated?

arjuntx2, 2020-07-21:
I will get back to you soon. Thanks alot

arjuntx2, 2020-07-24:
> Also, does it work when you run it as `detectnet.py`, but not `detectnet-camera.py`?
> 
> Can you try running it with the `--debug` flag added to get more info?

arjuntx2, 2020-07-24:
As You have mentioned in the document : 

_To classify some static test images, we'll use the extended command-line parameters to detectnet (or detectnet.py) to load our custom SSD-Mobilenet ONNX model. To run these commands, the working directory of your terminal should still be located in: jetson-inference/python/training/detection/ssd/_

If I go to this directory, In my case, there is no such a file named detectnet or detectnet.py in that directory

So I get this error : 

bash: detectnet: command not found

Soin that directory, is it still supposed to run the following command line? Am I making any mistake here?

detectnet --model=models/fruit/ssd-mobilenet.onnx --labels=models/fruit/labels.txt \
          --input-blob=input_0 --output-cvg=scores --output-bbox=boxes \
            "images/fruit_*.jpg" test_fruit

In my case I copied files _detectnet-console_ and _detectnet-camera_ from build/aarch64/bin/ and pasted it in jetson-inference/python/training/detection/ssd/

After that, I removed back slash as you suggested and I was able to create engine running detecnet-console

I trained it for Boy/Girl data and it is not detecting anything. 

**Error:detectnet-console:  writing 1067x1600 image to 'test_fruit'
[image] invalid extension format '.test_fruit' saving image 'test_fruit'
[image] valid extensions are:  JPG/JPEG, PNG, TGA, BMP, and HDR.
detectnet-console:  failed saving 1067x1600 image to 'test_fruit'*

Although, the example image has .jpg format.

About: sudo make install, in which directory am I supposed to run this command?

Thank you for answering again :)

dusty-nv, 2020-07-24:
> If I go to this directory, In my case, there is no such a file named detectnet or detectnet.py in that directory

When you do `sudo make install`, these get installed to `/usr/local/bin`, which means they should run from any directory.

You should run `sudo make install` from your `jetson-inference/build` directory.

> In my case I copied files _detectnet-console_ and _detectnet-camera_ from build/aarch64/bin/ and pasted it in jetson-inference/python/training/detection/ssd/

You could also do the reverse, run the programs from `jetson-inference/build/aarch64/bin`, and then adjust the paths to your custom model.   It doesn't actually need to be run from `jetson-inference/python/training/detection/ssd/`, it just makes the paths to your custom model shorter.

torabshaikh, 2020-07-26:
Hi @dusty-nv , is it possible to retrain the model for only two new classes on some other computer and use it on Jetson? I have tried using DIGITS locally but I kept running into one problem after another without running a single epoch.

dusty-nv, 2020-07-27:
Hi Torab, DIGITS is only supported on x86 (not Jetson), and the DIGITS portion of the Hello AI World tutorial is deprecated.

You can run the same PyTorch training code on PC running Ubuntu though (as long as PC has NVIDIA GPU).  Just install PyTorch and torchvision on your PC first.

drmnasr, 2020-08-10:
Hello,
@dusty-nv I actually tried this, i retrained on an EC2 instance, afterwards i took the pth file, converted it to onnx on a jetson nano and it didnt run, i got "killed" after "Graph construction and optimization completed in 0.103095 seconds." in the first run, couldn't create the engine. I tried the same exactly on my Xavier NX and that worked fine, the model isn't that large, 10 epochs on 2 classes of open images, batch size 128 though.

could that be the reason?

dusty-nv, 2020-08-10:
> batch size 128 though.

Yes, increased batch size will also increase the memory, so you may want to try small batch size for Nano.   Or mount swap on Nano for when it is building the TensorRT engine.

drmnasr, 2020-08-10:
@dusty-nv 
thank you for your reply, i tried again just 1 epoch with batch size 32 on the EC2 instance, then transferred to the nano and mounted swap 4G...same issue, i will try with batch size 12 and update you.

dusty-nv, 2020-08-10:
Try first with batch size 1, that is what I tested with.

dusty-nv, 2020-08-10:
Actually, the training batch size is independent of the ONNX batch size (that should always be 1).   So must be something else...

drmnasr, 2020-08-10:
then what can it be? it was a fresh install for the nano, works perfectly fine with the existing models for detectnet, and my custom model works fine on the xavier

dusty-nv, 2020-08-10:
Does it work with this model?   https://nvidia.box.com/shared/static/gq0zlf0g2r258g3ldabl9o7vch18cxmi.gz

Also, when you did a fresh install on the Nano, it was of JetPack 4.4, right?

drmnasr, 2020-08-10:
nope, just tested it, same thing, getting killed. just works with the pre-trained ones included in this repo, i tried 3 custom trained so far. and yes fresh install of 4.4

[image]

dusty-nv, 2020-08-10:
Can you run `sudo tegrastats` in another terminal to keep an eye on the memory usage?   'Killed' normally means out of memory... interesting, because it does load on my Nano here.

drmnasr, 2020-08-10:
seems to be out of memory yes, but why? the model is small and the pre-trained ones work

[image]

dusty-nv, 2020-08-10:
I'm not sure why it happens, since it doesn't run out of memory on my Nano.   Can you try disabling the Ubuntu UI and rebooting?

To disable GUI - https://askubuntu.com/a/1056371

Then with GUI disabled, try running on a single test image.   If it works, you can then re-enable the GUI because the TensorRT engine will already be built for next time.

drmnasr, 2020-08-10:
That actually worked! but for about 5 mins, then it is frozen in the tactic, i will leave it for a while, tegrastats shows 3300+ utilization, i will update you if it changes, if not, i will restart it

drmnasr, 2020-08-11:
I left it for hours, it is frozen in the middle of preparing the engine, anything else i can do to free more memory?

dusty-nv, 2020-08-11:
Hmm not sure why it isn't working for you.

Can you try running `sudo systemctl disable nvzramconfig`?   Then reboot (and remount your 4GB swap if you need to)

dusty-nv, 2020-08-13:
@drmnasr I think I found a way to work around this - comment out these lines:

https://github.com/dusty-nv/jetson-inference/blob/627b5890b49449573b7c1af8de22ce985fc395e4/c/detectNet.cpp#L112

```c++
//if( modelTypeFromPath(model) == MODEL_ONNX )
//		mWorkspaceSize = 2048 << 20;
```

Then re-run `make` and `sudo make install`

torabshaikh, 2020-08-14:
@drmnasr We were facing a similar problem with one of our Jetson Nano. I was using `detectnet.py` to load the model and it was taking hours to build the engine file and that too after failing multiple times. I switched to detectnet command from Python when loading the model first time. It built the engine file then switched to python to use the model.

drmnasr, 2020-08-16:
@dusty-nv I can never thank you enough. worked perfectly!! thank you!

Rafipaulino, 2021-05-03:
Hi @dusty-nv Has the original question in the post finally been resolved?
I am having similar problems....INVALID_ARGUMENT: Cannot find binding of given name: coverage.

Any idea about what could be happening?

dusty-nv, 2021-05-03:
> I am having similar problems....INVALID_ARGUMENT: Cannot find binding of given name: coverage.

This seems like a different issue.  It isn't getting the custom layers names.  Are you sure you have the following command line arguments included?  (check for typos)

`--input-blob=input_0 --output-cvg=scores --output-bbox=boxes`

Rafipaulino, 2021-05-04:
Oh sorry, my mistake!! You were right. It was a typo problem. Thanks!!!

closed 2023-04-27 by dusty-nv