dusty-nv/jetson-inference #650: detectnet python script example usage not working

silent-code, 2020-07-21:
I'd like to test the detectnet python script on a single image. The usage example on the documenation page for the object detection inference says to use the following command:

$ ./detectnet.py --network=ssd-mobilenet-v2 images/peds_0.jpg output.jpg 

but you get the error:

**Exception: jetson.utils -- failed to create videoSource device**

Is the usage documentation not up to date or am I using the wrong script to just test on a single image?

S4WRXTTCS, 2020-07-21:
This command works perfectly fine on my Jetson Xavier NX running the initial JetPack for the NX. I haven't upgraded yet to the most recent Jetpacks. 

Do the other Python Scripts work on your setup?

dusty-nv, 2020-07-21:
Hmm that should work... can you post your full console log please?

silent-code, 2020-07-21:
Issuing this command from the jetson-inference/python/examples/ directory works:

 ./detectnet.py --network=ssd-mobilenet-v2 /home/python_scripts/jetson-inference/data/images/peds_0.jpg output.jpg

Closing the issue. Thank you for your responses.

closed 2020-07-21 by silent-code