dusty-nv/jetson-inference #651: net.Detect - Anyway to change overlay "box"  into a "line bordered" shape

mmatesic01, 2020-07-23:
Is there a way to change "net.Detect(img, overlay=opt.overlay)" overlay "box" into a "line bordered" shape?
Thanks.

dusty-nv, 2020-07-23:
I initially tried this, but for some reason it was buggy drawing the lines in my CUDA kernels vs. filled rect.   So for now it is filled rect.

leandrovrabelo, 2020-07-23:
Why don't you try using OpenCV? something like this:

import cv2 

for i in range(len(detections)):

        img = cv2.rectangle(img, 
                                        (int(detections[i].Left), int(detections[i].Top)),
                                        (int(detections[i].Right), int(detections[i].Bottom)),
                                        detections[i].ClassID, 1)

dusty-nv, 2020-07-23:
cv2 doesn't use GPU, so I don't want to use it in the master codeline.

But you could skip the overlay (`overlay=none`) and render your own with cv2.

mmatesic01, 2020-07-24:
I already did that with cv2. 
I just hoped that I could eliminate cv2 out of the code and do the rectangle with jetson-inference.
Oh, well...

closed 2020-07-24 by mmatesic01

firststep-dev, 2023-01-21:
You can use the following to show an outline, instead of a solid fill box: `--overlay=line`

Overlay options to choose from include: `box`, `lines`, `labels`, `conf`, and `none`

Reference: https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-console-2.md