dusty-nv/jetson-inference #652: How do I use the retrained ssd-mobilenet.onnx model in a custom script?

silent-code, 2020-07-23:
How do I use the retrained ssd-mobilenet.onnx model in a custom script? 

Here's the code I have that works fine with the existing ssd-mobilenet-v1. It displays the video with the detections:

**network = "ssd-mobilenet-v1"
net = jetson.inference.detectNet(network=network, threshold=0.5)
video = './my_video.mp4'
cap = cv2.VideoCapture(video)
while (cap.isOpened()):
	ret, img = cap.read()
	img = cv2.rotate(img, cv2.ROTATE_90_CLOCKWISE)
	height = img.shape[0]
	width = img.shape[1]
	img = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)
	img = cv2.cvtColor(img, cv2.COLOR_RGB2RGBA).astype(np.float32)
	img = jetson.utils.cudaFromNumpy(img)
	detections = net.Detect(img, width, height)
	img = jetson.utils.cudaToNumpy(img, width, height, 4)
	img = cv2.cvtColor(img, cv2.COLOR_RGBA2RGB).astype(np.uint8)
	img = cv2.cvtColor(img, cv2.COLOR_RGB2BGR)
	cv2.imshow('detected image', img)
	if cv2.waitKey(1) & 0xFF == ord('q'):
		break**

I followed the steps to retrain the model and everything seems fine, however, I get the following error if I use network=ssd-mobilenet.onnx in the above script:

**jetson.inference -- detectNet loading build-in network '/home/tim/python_scripts/jetson-inference/python/training/detection/ssd/models/uuvs/ssd-mobilenet.onnx'
jetson.inference -- detectNet invalid built-in network was requested ('/home/tim/python_scripts/jetson-inference/python/training/detection/ssd/models/uuvs/ssd-mobilenet.onnx')
Traceback (most recent call last):
  File "my-detection.py", line 44, in <module>
    net = jetson.inference.detectNet(network=network, threshold=0.5)
Exception: jetson.inference -- detectNet invalid built-in network was requested**

Thanks in advance!

dusty-nv, 2020-07-23:
Hi @silent-code, normally this would be done from the command line like here:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md#processing-images-with-tensorrt

```bash
detectnet --model=models/fruit/ssd-mobilenet.onnx --labels=models/fruit/labels.txt \
          --input-blob=input_0 --output-cvg=scores --output-bbox=boxes \
            "images/fruit_*.jpg" test_fruit
```

However if you wish to hard-code these into your script, the easiest way would just be to construct it like this:

```python
net = jetson.inference.detectNet(argv=['--model=ssd-mobilenet.onnx', '--labels=models/fruit/labels.txt', '--input-blob=input_0', '--output-cvg=scores', '--output-bbox=boxes'])
```

Adjust the paths above for `--model` and `--labels` for your model.

silent-code, 2020-07-24:
That works! Thanks and sorry for nube question but I'm new to using python packages in custom code that have C++ bindings. The reason for wanting to go beyond the command line execution is to integrate the jetson-inference python module in an autonomous drone. Another final question. Why use arv[ ... ] in the net instantiation and not use something like the following 

```
net = jetson.inference.detectNet(model='../training/detection/ssd/models/uuvs/ssd-mobilenet.onnx',
							labels='../training/detection/ssd/models/uuvs/labels.txt',
							input-blob=input_0,
							output-cvg=scores, output-bbox=boxes, threshold=.15)
```

dusty-nv, 2020-07-24:
> Why use argv[ ... ] in the net instantiation and not use something like the following

The only reason is that I haven't written the bindings and argument parsing for those additional function arguments.   Since you can't overload the Python constructor, it makes the bindings complex to support 1) build-in model name, 2) command-line arguments, and 3) extended parameters

closed 2020-07-24 by silent-code

irfwas, 2020-11-01:
> Hi @silent-code, normally this would be done from the command line like here:
> 
> https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md#processing-images-with-tensorrt
> 
> ```shell
> detectnet --model=models/fruit/ssd-mobilenet.onnx --labels=models/fruit/labels.txt \
>           --input-blob=input_0 --output-cvg=scores --output-bbox=boxes \
>             "images/fruit_*.jpg" test_fruit
> ```
> 
> However if you wish to hard-code these into your script, the easiest way would just be to construct it like this:
> 
> ```python
> net = jetson.inference.detectNet(argv=['--model=ssd-mobilenet.onnx', '--labels=models/fruit/labels.txt', '--input-blob=input_0', '--output-cvg=scores', '--output-bbox=boxes'])
> ```
> 
> Adjust the paths above for `--model` and `--labels` for your model.

Hey @dusty-nv , 

Im using Jetpack 4.4 and I tried to implement the network in my script as you showed above but I get this error. 
However, when I run the "detectnet -model/models........." code in the terminal it works fine. This is my code and the error I receive

**### Code:**
`import jetson.inference
import jetson.utils

net = jetson.inference.detectNet(argv=['--model=models/adas/ssd-mobilenet.onnx', '--labels=models/adas/labels.txt', '--input-blob=input_0', '--output-cvg=scores', '--output-bbox=boxes'])
camera = jetson.utils.videoSource("csi://0")      # '/dev/video0' for V4L2
display = jetson.utils.videoOutput("display://0") # 'my_video.mp4' for file

while display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	display.Render(img)
	display.SetStatus("Forward Collision Detection {:.0f} FPS".format(net.GetNetworkFPS()))`

**### Output:**
`jetson.inference -- detectNet loading network using argv command line params
detectNet -- loading detection network model from:
          -- prototxt     NULL
          -- model        models/adas/ssd-mobilenet.onnx
          -- input_blob   'input_0'
          -- output_cvg   'scores'
          -- output_bbox  'boxes'
          -- mean_pixel   0.000000
          -- mean_binary  NULL
          -- class_labels models/adas/labels.txt
          -- threshold    0.500000
          -- batch_size   1
[TRT]    TensorRT version 7.1.3
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    detected model format - ONNX  (extension '.onnx')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    attempting to open engine cache file .1.1.7103.GPU.FP16.engine
[TRT]    cache file not found, profiling network model on device GPU
Error:  model file 'models/adas/ssd-mobilenet.onnx' was not found.
        if loading a built-in model, maybe it wasn't downloaded before.

        Run the Model Downloader tool again and select it for download:

           $ cd <jetson-inference>/tools
           $ ./download-models.sh
[TRT]    detectNet -- failed to initialize.
jetson.inference -- detectNet failed to load network
Traceback (most recent call last):
  File "/home/irfwas/Desktop/pypro/ForwardCollisionWarning/testRetrainedSSD.py", line 4, in <module>
    net = jetson.inference.detectNet(argv=['--model=models/adas/ssd-mobilenet.onnx', '--labels=models/adas/labels.txt', '--input_blob=input_0', '--output_cvg=scores', '--output_bbox=boxes'])
Exception: jetson.inference -- detectNet failed to load network `

dusty-nv, 2020-11-02:
> Error: model file 'models/adas/ssd-mobilenet.onnx' was not found.
> if loading a built-in model, maybe it wasn't downloaded before.

Hi @irfwas, are you running your script from the directory which contains `models/adas/ssd-mobilenet.onnx`?    

The current working directory of your terminal should be the directory above your `models/` folder.

As a quick test, you could also change your script to reference the absolute paths of your model and labels to make sure it can load first.

irfwas, 2020-11-03:
Thank you so much for your reply @dusty-nv. I was really looking forward to it.

Yes, I am trying to run the script from the directory which contains `models/adas/ssd-mobilenet.onnx`. I tried to train this model for 100 epochs but it got killed at around 52 epochs, so I decided to go ahead and convert it to an '.onnx' file. I tried to use it in the terminal using the `detectnet` as shown in the tutorial and it worked fine and even managed to hit 40 fps, yet I can't get it to work in the script.

I have tried to use the absolute paths for the models and labels as shown in the code below, however, I am still receiving the same error message as the message shown earlier. What should I do, or what am I doing wrong?

I really need your help as this is my final stage before having to present my project as part of my final year project (FYP) which is due in 2 weeks. I sincerely hope you can help. 

`net = jetson.inference.detectNet(argv=['--model=home/jetson-inference/python/training/detection/ssd/models/adas/ssd-mobilenet.onnx', '--labels=home/jetson-inference/python/training/detection/ssd/models/adas/labels.txt', '--input-blob=input_0', '--output-cvg=scores', '--output-bbox=boxes'])`