dusty-nv/jetson-inference #654: Reproducing detection results

deepadam, 2020-07-24:
hi  nvidia devs

I wanted to reproduce results of repo but in master branch i can not find the detectnet_camera.py script, what is the situation?

I had to clone depth branch, it has the scripts but when i run i get an error like  :    
```

File "detectnet-camera.py", line 58, in <module>
    img, width, height = camera.CaptureRGBA()
TypeError: 'jetson.utils.cudaImage' object is not iterable

```

Can you help me to solve this? Thank you

dusty-nv, 2020-07-25:
Hi, please clone the master branch and refer to the most recent docs from the tutorial.

The script is now just called 'detectnet.py', but when it gets installed a copy is made called 'detectnet-camera.py' and 'detectnet-console.py' for backwards compatability.  They are now the same script though.

deepadam, 2020-07-25:
Thank you Dusty, i wish you have a good day

 i will clone master branch, i am wondering can i use my custom onnx model that is trained using pytorch-ssd repository with master branch? I am using Jetson TX2 with jetpack 4.4 tensorRT 7.1

dusty-nv, 2020-07-25:
Yes, there was tutorial I added recently to cover this.  Please see "Re-training SSD-Mobilenet" page.

closed 2020-07-25 by deepadam