dusty-nv/jetson-inference #655: detectnet.py

g30ba1, 2020-07-25:
Hi there, I've installed jetson-inference several weeks ago, and I'm watching that there is an update for the master branch.

Do I need to install JetPack 4.4 for getting this updates running properly?

I'm trying to run detectnet.py with videos, but the error "jetson.utils doesn't have VideoSource attributtes"

Currently, I'm using JetPack 4.2 and I want to try the Object Detection (SSD-Mobilenet) features.

dusty-nv, 2020-07-26:
If you want to train object detection, then yes you would need to upgrade to JetPack 4.4.

g30ba1, 2020-07-27:
I see, but even if I don't to train object detection, I need JetPack 4.4 for being able to run inference using detecnet.py, am I right? 

Let me re-phrase: If I want to run detectnet.py inference, I need both JetPack 4.4 and clone (Or download the repository) again?

dusty-nv, 2020-07-27:
If you want to run detectnet inference on the PyTorch-trained SSD-Mobilenet ONNX model, then you need JetPack 4.4 on your Jetson.  This is because loading the SSD-Mobilenet ONNX model generated by PyTorch requires TensorRT 7.1.

If you don't care about loading that specific model, you can use older JetPack.   With older JetPack, you can load TensorFlow-trained SSD-Mobilenet (COCO model) that already comes with the repo.

closed 2023-04-27 by dusty-nv