dusty-nv/jetson-inference #656: Retraining MobileNetV2 on a highspec computer for new classes.

torabshaikh, 2020-07-26:
Hi.
Thanks for your amazing repo.

I have one issue while retraining the model. My Jetson device takes a lot of time and produces too much heat. Is it possible to retrain the model for only two new classes on some other computer and use it on Jetson? I have tried using DIGITS locally but I kept running into one problem after another without running a single epoch.

Thanks in advance.

dusty-nv, 2020-07-27:
Hi Torab, see my reply to your question in the other thread here:

https://github.com/dusty-nv/jetson-inference/issues/649#issuecomment-664066301

torabshaikh, 2020-08-14:
Trained MobileNetV1 using the [The other repo](https://github.com/dusty-nv/pytorch-ssd) and the model is running perfectly on Jetson Nano with average 35 FPS. However, there is a little compromise with accuracy for which we are building a custom dataset to further train it on. Thanks for your help.

closed 2020-08-14 by torabshaikh