dusty-nv/jetson-inference #657: Where to change dataset type on camera-capture?

AiueoABC, 2020-07-27:
I read the document to make a dataset of mobilenetv2-ssd. 

When I tried to do this following "jetson-inference/docs/pytorch-collect-detection.md", I couldn't find an option to change dataset type on "Data Capture Control" after executing "camera-capture --camera /dev/video0". It seems the appearance is different.

What I tried first is to update this repo, since I cloned this for long time ago, but this didn't work.

Do I miss something?

dusty-nv, 2020-07-27:
Yes, tool was updated recently to support it.

You might want to reclone the repo with `--recursive` flag to make sure you are using the latest.

AiueoABC, 2020-07-28:
Thanks dusty!
I was stupid to forget that...

dusty-nv, 2020-07-28:
No worries @AiueoABC - let me know if you run into any other problems using the tool.

closed 2020-07-29 by AiueoABC