dusty-nv/jetson-inference #658: Compiling error: gst_structure_get_list

alexcablo, 2020-07-27:
Hi, I'm following the instructions of Build the Project from Source, but I'm getting the following error while compiling ($ make): 

gst_structure_get_list’ was not declared in this scope

any idea in how to fix it? Thanks!!

dusty-nv, 2020-07-27:
@alexcablo , please clone the `20200714` branch for older support:

https://github.com/dusty-nv/jetson-inference/tree/20200714

shabaz-git, 2020-08-26:
Hi @alexcablo Did your issue resolved?
Because I am also getting the same error and I am unable to clone the 20200714 branch :

$ (venv_base) nvidia@tegra-ubuntu:~/shabaz/temp$ git clone --recursive https://github.com/dusty-nv/jetson-inference/tree/20200714
Cloning into '20200714'...
fatal: repository 'https://github.com/dusty-nv/jetson-inference/tree/20200714/' not found

And I have checked the changes between this branch and the master branch, there are no changes.
My jetpack version is also same as yours.What is the solution for this?

@dusty-nv you can help?

sahiustatt, 2020-09-23:
> Hi, I'm following the instructions of Build the Project from Source, but I'm getting the following error while compiling ($ make):
> 
> gst_structure_get_list’ was not declared in this scope
> 
> Jetpack 3.3 | 3.2.1 [L4T 28.2.1]
> 
> any idea in how to fix it? Thanks!!

Hey did the issue resolved. I am having something similar.

Yao1up, 2020-09-26:
> Hey did the issue resolved. I am having something similar.

did you issue it , I have the same problem.

Meschmid23, 2020-09-27:
I have the same problem. 

I tried to checkout branch 20200714, and ran `make -j$(nproc)` and again the the same error as the OP.

shabaz-git, 2020-10-12:
Hi @Meschmid23 
For me, 20200714 branch is working fine. 
For me, the make error with the master branch is not resolved.

I followed these steps :

$ git clone --single-branch --branch 20200714 --recursive https://github.com/dusty-nv/jetson-inference
$ cd jetson-inference/

# check for the branch details
$ git branch
   * 20200714

$ mkdir build
$ cd build/
$ cmake ../
$ make -j$(nproc)
# if it didn't work then try only make
  $ make
$ sudo make install
$ sudo ldconfig

If anyone got the solution for the master branch **make** issue please provide the solution.

thank you

wuzhishiwo, 2020-11-09:
hello, I also meet the same problem,I am a green hand,could you give the detailed instructions to solve it. I  am really looking forward to your reply and help,thank you very much.

Yao1up, 2020-11-09:
> hello, I also meet the same problem,I am a green hand,could you give the detailed instructions to solve it. I am really looking forward to your reply and help,thank you very much.
I had cloned the 20200714 branch and the code works.

dusty-nv, 2020-11-09:
Yes, if you clone the `20200714` branch it will work:

``` bash
$ git clone --recursive -b 20200714 https://github.com/dusty-nv/jetson-inference
```

Although ideally you would upgrade your JetPack to something more recent (JetPack 4.4.1 is the latest), which will also update GStreamer and make the build error in master branch go away.

Meschmid23, 2020-11-09:
Upgrading to the most recent JetPack  (4.4.1 at the time of writing) solved the issue for me.

I was worried about the backwards compatibility for my specific use case, but everything worked out and there where no issues after upgrading.

closed 2023-04-27 by dusty-nv