dusty-nv/jetson-inference #659: How do I perform Transfer Learning on the EfficientNet model series?

gapro20022, 2020-07-27:
I want to train a pre-trained model with my own dataset. I've already done this with the available models from the project. However, I want to try training other new models, specifically the EfficientNet series, to compare the performance and accuracy of such models. Is there a way to achieve this?

Thank you very much,
Henry

dusty-nv, 2020-07-27:
Hi Henry, I don't believe EfficientNet is one of the network architectures supported in torchvision, so you would need to use a custom PyTorch project that supported it, like one of these perhaps:

- https://github.com/lukemelas/EfficientNet-PyTorch
- https://github.com/rwightman/pytorch-image-models/

You would also need to export the model from PyTorch to ONNX format.

gapro20022, 2020-07-27:
Thank you! I will check those two out.

closed 2020-07-27 by gapro20022