dusty-nv/jetson-inference #666: How to save video processing on detectnet

Sgomes14843, 2020-07-30:
Hello, I am using this code so that my model can process video instead of just images on the detectnet, it works but I can only see it while processing. How do I get him to record this video with the bounding boxes and results in an .mp4 file?
[image]

dusty-nv, 2020-07-31:
Hi @Sgomes14843 , add `videos/my_output.mp4` to the end of your command line.

Sgomes14843, 2020-07-31:
Hi
[image]
, first thanks for responding.
I tried this but the output is like a text file instead of video. Do you know what it might be?

Sgomes14843, 2020-07-31:
> Hi @Sgomes14843 , add `videos/my_output.mp4` to the end of your command line.
Hi, first thanks for responding.
I tried this but the output is like a text file instead of video. Do you know what it might be?
[image]

dusty-nv, 2020-07-31:
I think it loos like the text file icon shortly after the video has been created, before Ubuntu recognizes it.

If you try to open the video in a video player, does it play?

Sgomes14843, 2020-07-31:
> I think it loos like the text file icon shortly after the video has been created, before Ubuntu recognizes it.
> 
> If you try to open the video in a video player, does it play?

If you open it in the video player it doesn't give and gives this error and if it opens as a notepad it is empty, there is nothing in the file
[image]

dusty-nv, 2020-07-31:
Can you provide the command line that you ran along with the terminal output?

Did the process shutdown gracefully when you either closed the GUI window or pressed Ctrl+C?

Sgomes14843, 2020-07-31:
The command line:
`python3 detectnet.py --model=models/cars/ssd-mobilenet.onnx --labels=models/cars/labels.txt \
          --input-blob=input_0 --output-cvg=scores --output-bbox=boxes \
            videos/f1.mp4   videos/my_output.mp4`
Yes the process closed properly when i close the GUI or press CRTL + C
Here's what it says on the terminal when I close the GUI or press CTRL + C

dusty-nv, 2020-07-31:
Hmm, well it appears to be running correctly.  If you run `ls -ll videos/my_output.mp4`, does it also show a size of 0 bytes?

Sgomes14843, 2020-07-31:
Yes
[image]

dusty-nv, 2020-07-31:
I can't seem to tell why it's doing that, sorry.   Can you try removing the backslashes (`\`) from your command line to see if that makes a difference:

```bash
python3 detectnet.py --model=models/cars/ssd-mobilenet.onnx --labels=models/cars/labels.txt  --input-blob=input_0 --output-cvg=scores --output-bbox=boxes  videos/f1.mp4 videos/my_output.mp4
```

Also can you try running the C++ version of the application?

```bash
detectnet --model=models/cars/ssd-mobilenet.onnx --labels=models/cars/labels.txt  --input-blob=input_0 --output-cvg=scores --output-bbox=boxes  videos/f1.mp4 videos/my_output.mp4
```

Failing that, please post the full text log of the programming running so I can look it over - thanks.

Sgomes14843, 2020-07-31:
Unfortunately none of these lines of code worked.
How do i do the full text log of the programming running ?

dusty-nv, 2020-07-31:
You can either copy it from your terminal into a text file, or launch the program like this with `| tee log.txt` to write the log:

```bash
detectnet --model=models/cars/ssd-mobilenet.onnx --labels=models/cars/labels.txt  --input-blob=input_0 --output-cvg=scores --output-bbox=boxes  videos/f1.mp4 videos/my_output.mp4 | tee log.txt
```

Sgomes14843, 2020-07-31:
Thanks for teaching how to do this, here's the log file
[log.txt](https://github.com/dusty-nv/jetson-inference/files/5008723/log.txt)

dusty-nv, 2020-07-31:
Thank you, I didn't identify any errors that jumped out at me.   But maybe try saving a different video format - `my_output.mkv` instead of `my_output.mp4`.   Does that help?

Also, it occurred to me to try a more regular input resolution instead of 854x480, like 640x480 (just as a test):

```bash
detectnet --model=models/cars/ssd-mobilenet.onnx --labels=models/cars/labels.txt  --input-blob=input_0 --output-cvg=scores --output-bbox=boxes --input-width=640 --input-height=480 videos/f1.mp4 videos/my_output.mkv
```

Sgomes14843, 2020-07-31:
It worked, yeah!!!
Thank you so much for all your help and the time you spent
And I tested it with .mp4 and it worked so the problem was the resolution.

dusty-nv, 2020-07-31:
Aha, ok - glad you got it working.  That is strange, is seems the encoder didn't like that resolution but didn't output any messages/warnings about it.  I will have to remember that...

closed 2020-07-31 by Sgomes14843