dusty-nv/jetson-inference #669: Error while running my-detection.py

tanueihorng, 2020-08-04:
Hi! 

Anyone knows why i got this error while runing the python script ?
How can i solve it ? Thanks a lot!!!

[image]
[image]
[image]
[image]

[FULLerror.txt](https://github.com/dusty-nv/jetson-inference/files/5019820/FULLerror.txt)

`import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-inception-v2", threshold=0.4)
camera = jetson.utils.videoSource("rtsp://root:leadSingtel@192.168.8.120:554/axis-media/media.amp", argv=['--input-codec=h264'])
display = jetson.utils.videoOutput("display://0") # 'my_video.mp4' for file

while display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	display.Render(img)
	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
`

Rbara00, 2020-09-10:
Ive been having the same exact problem, did you figure it out?

feimiao888, 2020-12-07:
Refer to #619, a temporary workaround is to move **import jetson.inference** after the line jetson.utils.videoOutput

ngzhongcai, 2021-09-19:
Is this issue being worked at? I am trying to crop out detected images, and convert them to numpy arrays, using the jetson.utils methods. Above temporary solution precludes me from doing so.

closed 2023-04-27 by dusty-nv