dusty-nv/jetson-inference #671: Training of my own classification model with very bad results (100 epochs)

Sgomes14843, 2020-08-05:
Hi, I trained my own dataset following the tutorial "Re-training on the PlantCLEF Dataset". My dataset consists of 10 classes of F1 cars per team. So I trained following the steps of the tutorial, with 100 epochs and batch size of 16. I used resnet50 and then resnet 18. With the onnx model I tested on jetson and the results are bad. The classifier only hits me only one class of 10, otherwise it changes everything, for example saying that a ferrari is a mercedes and etc ...
In the images I attached, he only hit the toro_rosso, and it seems to me that it is the only class that he hits right.
Did the problem have been training with 100 epochs, is more needed?

[image]
[image]
[image]

dusty-nv, 2020-08-05:
Hi @Sgomes14843 , what was the accuracy of the model in PyTorch after epoch 100?   How many images per class were in your training dataset?

Also, is your `labels.txt` file in alphabetical order?

Sgomes14843, 2020-08-05:
Hi, I don't remember well but I think an ACC@1 was between 75 to 90.
The number of images, each class in the train folder has between 700 to 800 images. And my labels are not in alphabetical order, they are like this:
[image]

dusty-nv, 2020-08-05:
You need to change the labels to alphabetical order   (the way they would be listed if you did `ls` on your dataset dir)

This is because pytorch doesn't use labels file, rather it loads the class directories in alphanumeric order.

Sgomes14843, 2020-08-05:
Yes, I already tried it and it worked
Thank you for your help!!

closed 2020-08-05 by Sgomes14843

dusty-nv, 2020-08-05:
OK great, glad that you got it working!