dusty-nv/jetson-inference #674: own detection datasets -Error  too many indices

michalrusinowski, 2020-08-08:
Dear, 

Thank you for the tutorial.
I successfully done mosto of the exaples that you procided.
Now I am on 

https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md

And I keep getting

An error.

So what I've done, 
1. I created my own dataset as you described in the tutorial, 
2. I run python3 ../train_ssd.py --dataset-type=voc --data=dataset (I have this in separate project folder therefore I have "../train_ssd.py ")

**The things I tried to do after the error occured**

3. I tried to change the folder - I thought this is the problem
4. I tried to copy the pictures to the test/train/val folders (originally it is in JPEGImages only)
5. I've seen some problems in the forum but it seams the other problems also gives too many indices but for other reason.
6. As you've seen I haven't use specific model - it got a new one.
7. I tried to avoid doing changes in the code...

Please help...

The error that I got is :

> 2020-08-08 15:50:22 - Start training from epoch 0.
/home/michal/.local/lib/python3.6/site-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
/home/michal/jetson-inference/python/training/detection/ssd/vision/transforms/transforms.py:247: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray
  mode = random.choice(self.sample_options)
/home/michal/jetson-inference/python/training/detection/ssd/vision/transforms/transforms.py:247: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray
  mode = random.choice(self.sample_options)
/home/michal/.local/lib/python3.6/site-packages/torch/nn/_reduction.py:44: UserWarning: size_average and reduce args will be deprecated, please use reduction='sum' instead.
  warnings.warn(warning.format(ret))
Traceback (most recent call last):
  File "../train_ssd.py", line 343, in <module>
    device=DEVICE, debug_steps=args.debug_steps, epoch=epoch)
  File "../train_ssd.py", line 113, in train
    for i, data in enumerate(loader):
  File "/home/michal/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 354, in __next__
    data = self._next_data()
  File "/home/michal/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 980, in _next_data
    return self._process_data(data)
  File "/home/michal/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 1005, in _process_data
    data.reraise()
  File "/home/michal/.local/lib/python3.6/site-packages/torch/_utils.py", line 395, in reraise
    raise self.exc_type(msg)
IndexError: Caught IndexError in DataLoader worker process 1.
Original Traceback (most recent call last):
  File "/home/michal/.local/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 185, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/michal/.local/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/michal/.local/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/michal/.local/lib/python3.6/site-packages/torch/utils/data/dataset.py", line 207, in __getitem__
    return self.datasets[dataset_idx][sample_idx]
  File "/home/michal/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 64, in __getitem__
    image, boxes, labels = self.transform(image, boxes, labels)
  File "/home/michal/jetson-inference/python/training/detection/ssd/vision/ssd/data_preprocessing.py", line 34, in __call__
    return self.augment(img, boxes, labels)
  File "/home/michal/jetson-inference/python/training/detection/ssd/vision/transforms/transforms.py", line 55, in __call__
    img, boxes, labels = t(img, boxes, labels)
  File "/home/michal/jetson-inference/python/training/detection/ssd/vision/transforms/transforms.py", line 99, in __call__
    boxes[:, 0] /= width
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed

Segmentation fault (core dumped)

Slade8566, 2020-08-18:
Did you fix it ?

Slade8566, 2020-08-18:
Hey i got it.
Change class names. ssd/vision/datasets/voc_dataset.py line 46.
Look this: https://github.com/qfgaohao/pytorch-ssd/issues/15

michalrusinowski, 2020-08-19:
Günaydın @Slade8566  

 Well I did... In my case there was some more problems and let me just go thru general problems that I had withdataset :
1. First of all there is no bug reporting so you either look at file that you have problem with and just change the script so you know on which annotation file it works and which file does it read or you will have some problems.
2. When you label the pictures, never save a file without labels (don't save the file if you did not draw a rectangle on it). The system reads the annotation and it crshes because it expects some data (list) but it has none. So the way you may check it is you simply go to annotations folder and look at size of the files - if one of them has half of the size of others than you have bug. Than what you need to do is to delete the name of  the picture connected with this annotation file from the lists of the files (/imageSets/Main/{test,train,trainval,val}).
3. image files may not include special chars and preferrably no spaces. 
4. file extention needs to be jpg not jpeg. remember if you change it after labeling than you need to change the names in annotation files (it is easy to do  in bash ).
5. If you're starting with LabelImage (if you want to start with saved pictures not with camera to grub the pic. )  than I would strongly recommend to start camera-capture , save one annotation and look at the folders and files structure that it created and do same thing in LabelImage.

I had also some more problems but I simply changed the python script so it showed me on which picture or annotation file it is now so I knew where to find the problem...

closed 2020-08-19 by michalrusinowski