dusty-nv/jetson-inference #676: module 'jetson.utils' has no attribute 'loadImage'

JRyan1234, 2020-08-12:
Hello!

I've been really enjoying the step by step instructions on how to get things started with the Jetson, really helps me learn. I was following along with the Image Recognition in Python example but keep running into the following issue when I run the program:

test@test-desktop:~/my-recognition-python$ ./my-recognition.py brown_bear.jpg 
jetson.inference.__init__.py
jetson.inference -- initializing Python 3.6 bindings...
jetson.inference -- registering module types...
jetson.inference -- done registering module types
jetson.inference -- done Python 3.6 binding initialization
jetson.utils.__init__.py
jetson.utils -- initializing Python 3.6 bindings...
jetson.utils -- registering module functions...
jetson.utils -- done registering module functions
jetson.utils -- registering module types...
jetson.utils -- done registering module types
jetson.utils -- done Python 3.6 binding initialization
Traceback (most recent call last):
  File "./my-recognition.py", line 16, in <module>
    img = jetson.utils.loadImage(opt.filename)
AttributeError: module 'jetson.utils' has no attribute 'loadImage'

I've tried the sudo make install and sudo ldconfig but had no changes. Any assistance you could offer would be great. 

Thanks much

dusty-nv, 2020-08-13:
Hmm, I wonder if it's possible that you are running an older version of the jetson-inference library?

Can you run this:

```bash
cd jetson-inference
git pull origin master
cd build
cmake ../
make
sudo make install
```

JRyan1234, 2020-08-14:
Hi Dusty! 

Thanks so much for the suggestion. I ran the commands you suggested but bumped into some errors:

In cmake ../:
......
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
CMake Warning at tools/camera-capture/CMakeLists.txt:8 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.

-- camera-capture:  Qt5Widgets package not found, skipping...

......

And then when I ran make to see if it would work despite the above error, I ran into this:

[ 35%] Linking CXX shared library ../aarch64/lib/libjetson-utils.so
[ 35%] Built target jetson-utils
[ 36%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_tensorConvert.cu.o
In file included from /home/test/jetson-inference/c/tensorConvert.cu:23:0:
/home/test/jetson-inference/c/tensorConvert.h:28:10: fatal error: imageFormat.h: No such file or directory
 #include "imageFormat.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
CMake Error at jetson-inference_generated_tensorConvert.cu.o.cmake:219 (message):
  Error generating
  /home/test/jetson-inference/build/CMakeFiles/jetson-inference.dir/c/./jetson-inference_generated_tensorConvert.cu.o

CMakeFiles/jetson-inference.dir/build.make:504: recipe for target 'CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_tensorConvert.cu.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_tensorConvert.cu.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Any thoughts? It might well be something very simple I'm doing wrong but am having trouble seeing it.

JRyan1234, 2020-08-14:
Hey Dusty, 

I read further in the cmake ../ error and found this line: "sudo apt-get install qtbase5-dev." Ran it and then re-ran the rest of the commands and it seems to be working fine now. Thanks so much for the help. Next challenge is feeding in video.

dusty-nv, 2020-08-14:
OK great - yea, I was going to suggest just doing a fresh re-clone of the latest master in case it didn't work.

closed 2023-04-27 by dusty-nv