dusty-nv/jetson-inference #677: Inference using ONNX model with and without TRT

AiueoABC, 2020-08-14:
Hi,
I managed to make my own ONNX model and run it with TRT following  "jetson-inference/docs/pytorch-collect-detection.md".
(It was really fun to make my own model and run it, thank you for this usuful repo!)

Now I'm curious about inference speed difference between with TRT and without TRT in my environment.
So I want to know how to run this generated ONNX model directly from python without using TRT.

Is there anyone tried this?

What I found is to use ONNXRuntime in python, but I couldn't understand how I can use this with my ONNX model generated here...

AiueoABC, 2020-08-18:
This is the code not working correctly, it returns boxes and scores, but never returns correct results seen in jetson-inference.

```
import onnxruntime
import numpy as np
import cv2

def main(cam):
    # Load image
    ret, frame = cam.read()
    width, height = 300, 300
    frame = cv2.resize(frame, (300, 300))

    # HWC to NHWC
    image_data = np.expand_dims(frame.transpose((2, 0, 1)), 0).astype(np.float32)

    # 1. Make session
    session = onnxruntime.InferenceSession('./model/ssd/mb2-ssd-lite.onnx')

    # 2. Get input/output name
    input_name = session.get_inputs()[0].name   # 'image'
    output_name_scores = session.get_outputs()[0].name     # 'scores'
    output_name_boxes = session.get_outputs()[1].name   # 'boxes'

    # 3. Run
    outputs_index = session.run([], {input_name: image_data})

    # Result
    output_scores = outputs_index[0]
    output_boxes = outputs_index[1]

    # Check input/output name
    print(input_name, output_name_boxes, output_name_scores)

    for scores in output_scores[0]:
        max_index = np.argmax(scores)
        if max_index != 0:
            print(max_index)
            boxes = output_boxes[0][max_index]
            top = boxes[0] * height
            left = boxes[1] * width
            bottom = boxes[2] * height
            right = boxes[3] * width

            top = max(0, np.floor(top + 0.5).astype('int32'))
            left = max(0, np.floor(left + 0.5).astype('int32'))
            bottom = min(height, np.floor(bottom + 0.5).astype('int32'))
            right = min(width, np.floor(right + 0.5).astype('int32'))

            cv2.rectangle(frame, (left, top), (right, bottom), (255, 255, 0), 4)
        cv2.imshow('annotated', frame)

if __name__ == '__main__':
    cap = cv2.VideoCapture(0)
    while True:
        main(cap)
        if cv2.waitKey(0) & 0xff == ord('q'):
            break
    cv2.destroyAllWindows()
    cap.release()
```
They make warnings a lot so maybe this is the reason, but not sure how to fix. 
What I got before inference is this;
```
2020-08-17 19:16:44.2335104 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.16.conv.1.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2341353 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.18.1.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2347130 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.0.1.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2351802 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.17.conv.7.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2356974 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.1.conv.1.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2361732 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.12.conv.4.num_batches_tracked'. It is not used by any node and should be removed from the model.
[... 45 lines cut ...]
2020-08-17 19:16:44.2604859 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.4.conv.1.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2610230 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'extras.3.conv.1.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2614735 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.4.conv.4.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2620125 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.4.conv.7.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2624624 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.5.conv.4.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2629851 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.6.conv.1.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2634761 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.6.conv.4.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2640517 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.6.conv.7.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2645179 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'base_net.7.conv.7.num_batches_tracked'. It is not used by any node and should be remo [... cut ...]
2020-08-17 19:16:44.2710270 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'regression_headers.0.1.num_batches_tracked'. It is not used by any node and should be removed from the model.
2020-08-17 19:16:44.2715208 [W:onnxruntime:, graph.cc:2709 onnxruntime::Graph::CleanUnusedInitializers] Removing initializer 'regression_headers.2.1.num_batches_tracked'. It is not used by any node and should be removed from the model.
```
Layers seemed to be removed?

douglastehling, 2021-01-22:
Hello! I wanted to take a doubt that I didn't understand. why does the code force you to use the 300x300 size? another question is, within the detected boxes, how can i receive the object ID to relate to a label.txt?

AiueoABC, 2021-01-22:
Hi,
why I used 300x300 image is bc I noticed my onnx model accept that size
when I see the model using Netron app.

And, for the label, I couldn't get anything from the model but found out
the index of the list returned is important to find the label of boxes. I
noticed I can find correct label when I use that number and label.txt
together.

dusty-nv, 2021-01-22:
> why I used 300x300 image is bc I noticed my onnx model accept that size when I see the model using Netron app.

@AiueoABC @douglastehling that is correct, this is the resolution that the model was trained at.  For example, the default training resolution for ResNet/GoogleNet/Alexnet is 224x224.  Inception is 299x299.  SSD is 300x300.

In theory, you could change these in the training scripts before training the models, but I haven't tested it with alternative resolutions.

AiueoABC, 2021-01-27:
@douglastehling 
This is how the values look like inside scores. In this case, max_index in my code will be '2', and that should be objectID.
[image]