dusty-nv/jetson-inference #679: Use fine grain jpeg data set in place of using camera for data set capture

TCIII, 2020-08-15:
Dusty,
This is really not an issue, but a request and might be of interest to other users with a similar need.

I have access to a fine grain aircraft jpeg data set that has 10,200 jpeg photos with 100 images each of 102 aircraft.
The images are stored as: data/images/1187707.jpg
The labels (list of training images) are stored as: data/images_train.txt

Would it be possible to use the fine grain images and their accompanying labels in place of using a camera to record the data set of interest?
My problem is understanding the CLI arguments structure necessary to train the model since I didn't have to use the Data Capture Control tool and might not have the same file structure which will need to be modified.

Your appearance on Paul McWhorter's Shop Talk on Friday was very entertaining and informative.

Regards,
TCIII

dusty-nv, 2020-08-17:
Hi @TCIII, the PyTorch classification training program expects the datasets to have the following directory structure:

```
‣ train/
	• class-A/
	• class-B/
	• ...
‣ val/
	• class-A/
	• class-B/
	• ...
‣ test/
	• class-A/
	• class-B/
	• ...
```

That is, the images are sorted into different sub-folders based on their class.

What I would do, is write a simple Python or bash script that reads your `images_train.txt` file, and copies the images to the appropriate subfolders.   On the other hand, you could write your own PyTorch dataloader, but it probably be easier just to sort the images into different folders as described above.

dusty-nv, 2020-08-17:
And thanks for tuning into the livestream last week - that was fun!  :)

TCIII, 2020-08-17:
Hi Dusty,
Thanks for the very informative response, much appreciated.
Okay, I will give it shot and close this issue.
Regards,
TCIII

closed 2020-08-17 by TCIII