dusty-nv/jetson-inference #680: imagnet fails to load

JVTEAM, 2020-08-17:
[TRT]   coreReadArchive.cpp (38) - Serialization Error in verifyHeader: 0 (Version tag does not match)
[TRT]   INVALID_STATE: std::exception
[TRT]   INVALID_CONFIG: Deserialize the cuda engine failed.
[TRT]   device GPU, failed to create CUDA engine
[TRT]   failed to load networks/bvlc_googlenet.caffemodel
[TRT]   imageNet -- failed to initialize.
imagenet-console:   failed to initialize imageNet

dusty-nv, 2020-08-17:
Hi @JVTEAM , delete your `bvlc_googlenet.*.engine` file and re-run - it should be found under:   `jetson-inference/data/networks`

For example, mine is `jetson-inference/data/networks/bvlc_googlenet.caffemodel.1.1.7103.GPU.FP16.engine`

Delete that, and then try running the program again.  It will re-build the engine for your updated version of TensorRT.

JVTEAM, 2020-08-17:
That worked.  What needs to be deleted for detectnet?

dusty-nv, 2020-08-17:
It depends on which model you are using.  Go into the models subfolder under `jetson-inference/data/networks/*` and delete all the `.engine` files.

JVTEAM, 2020-08-17:
There's only one .engine and that's the one that just got rebuilt.  Detectnet and segnet are failing in similar fashion.

closed 2020-08-18 by JVTEAM