dusty-nv/jetson-inference #681: Run Python Modules With Pednet

JVTEAM, 2020-08-17:
I'm trying to run detectnet_camera.py with pednet.  Doesn't seem to want to run.

python3 detectnet-camera.py --model ~/jetson-inference/data/networks/ped-100/snapshot_iter_70800.caffemodel --network pednet

[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - custom  (extension '.')
[TRT]   model format 'custom' not supported by jetson-inference
detectNet -- failed to initialize.
jetson.inference -- detectNet failed to load built-in network 'pednet'
PyTensorNet_Dealloc()
Traceback (most recent call last):
  File "detectnet-camera.py", line 49, in <module>
    net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)
Exception: jetson.inference -- detectNet failed to load network

dusty-nv, 2020-08-17:
What happens when you run `python3 detectnet-camera.py --network=pednet`?

Does it work when you run the C++ version?   `detectnet-camera --network=pednet`

JVTEAM, 2020-08-17:
Same thing happens.  The CPP works.  

I think if I just modify for the my-detection.py, that'll work better.

closed 2020-08-18 by JVTEAM