dusty-nv/jetson-inference #682: Segmentation fault (core dumped) on importing jetson-inference

macaquedev, 2020-08-17:
Dear Dusty,
 
I have a problem when I am importing jetson-inference. I flashed a 32GB card around 1 month ago for my jetson nano device. I then realised I am running out of space and upgraded to a 128GB. I installed all of the packages again from scratch, but then found that Jetson Inference is broken. It wasn't too happy when compiling too, telling me about a load of syntax errors...

Any help would be much appreciated.

dusty-nv, 2020-08-17:
Hi @macaquedev , had you flashed the latest JetPack 4.4 image to your 128GB SD card?

You shouldn't see compiling errors with that.  Were you using the jetson-inference master branch?

macaquedev, 2020-08-17:
Dear Dusty,

Wow that was a quick reply! Yes I had. I used Paul McWhorter's tutorial on it. I git cloned it using the command:
`git clone --recursive https://github.com/dusty-nv/jetson-inference`

dusty-nv, 2020-08-17:
Are you able to do a quick test by running the C++ version of the image recognition app?

```bash
cd path/to/your/jetson-inference/build/aarch64/bin
./imagenet images/jellyfish.jpg test_jellyfish.jpg
```

macaquedev, 2020-08-17:
It's training... hope it doesn't mess up...

dusty-nv, 2020-08-17:
Ah if it's training, yea don't try to run this also.  May run out of memory

macaquedev, 2020-08-17:
Sorry, what do you mean? Should I wait for it to train and then try again?
By the way, it finished. It displayed about half of the image and then gave me an error.

`imageLoader -- End of Stream has been reached...
imagenet: failed to capture next frame`

macaquedev, 2020-08-17:
Dear @dusty-nv,

Also, I was looking for an older release of jetson-inference but I can't seem to find anything on GitHub (there seems to be only one release).

macaquedev, 2020-08-18:
Dear @dusty-nv,

I have solved the problem. I just tried to reinstall jetson-inference using your Hello AI World tutorial and everything worked fine. Thank you very much for your help!

dusty-nv, 2020-08-18:
> By the way, it finished. It displayed about half of the image and then gave me an error.

Did it output the correct image to `test_jellyfish.jpg`?

dusty-nv, 2020-08-18:
> I have solved the problem. I just tried to reinstall jetson-inference using your Hello AI World tutorial and everything worked fine. Thank you very much for your help!

@macaquedev OK great, glad you were able to get it working!

macaquedev, 2020-08-18:
No it didn't output the correct image, but now after all tests it is working.

closed 2020-08-18 by macaquedev