dusty-nv/jetson-inference #683: Install error makefile:117:

yangs12, 2020-08-18:
I was running the code "sudo make install" and encountered the following error. What should I do next to fix it? Any help is appreciated. Thank you.

Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/local/include/jetson-inference/detectNet.h
-- Up-to-date: /usr/local/include/jetson-inference/imageNet.h
-- Up-to-date: /usr/local/include/jetson-inference/segNet.h
-- Up-to-date: /usr/local/include/jetson-inference/tensorConvert.h
-- Up-to-date: /usr/local/include/jetson-inference/tensorNet.h
-- Up-to-date: /usr/local/include/jetson-inference/randInt8Calibrator.h
-- Up-to-date: /usr/local/lib/libjetson-inference.so
-- Up-to-date: /usr/local/share/jetson-inference/cmake/jetson-inferenceConfig.cmake
-- Up-to-date: /usr/local/share/jetson-inference/cmake/jetson-inferenceConfig-noconfig.cmake
-- Up-to-date: /usr/local/bin/imagenet
-- Up-to-date: /usr/local/bin/imagenet-camera
-- Up-to-date: /usr/local/bin/imagenet-console
-- Up-to-date: /usr/local/bin/detectnet
-- Up-to-date: /usr/local/bin/detectnet-camera
-- Up-to-date: /usr/local/bin/detectnet-console
-- Up-to-date: /usr/local/bin/segnet
-- Up-to-date: /usr/local/bin/segnet-camera
-- Up-to-date: /usr/local/bin/segnet-console
-- Up-to-date: /usr/local/include/jetson-utils/URI.h
-- Up-to-date: /usr/local/include/jetson-utils/XML.h
-- Up-to-date: /usr/local/include/jetson-utils/alphanum.h
-- Up-to-date: /usr/local/include/jetson-utils/commandLine.h
-- Up-to-date: /usr/local/include/jetson-utils/csvReader.h
-- Up-to-date: /usr/local/include/jetson-utils/csvWriter.h
-- Up-to-date: /usr/local/include/jetson-utils/filesystem.h
-- Up-to-date: /usr/local/include/jetson-utils/logging.h
-- Up-to-date: /usr/local/include/jetson-utils/mat33.h
-- Up-to-date: /usr/local/include/jetson-utils/pi.h
-- Up-to-date: /usr/local/include/jetson-utils/rand.h
-- Up-to-date: /usr/local/include/jetson-utils/timespec.h
-- Up-to-date: /usr/local/include/jetson-utils/csvReader.hpp
-- Up-to-date: /usr/local/include/jetson-utils/csvWriter.hpp
-- Up-to-date: /usr/local/include/jetson-utils/gstCamera.h
-- Up-to-date: /usr/local/include/jetson-utils/v4l2Camera.h
-- Up-to-date: /usr/local/include/jetson-utils/gstDecoder.h
-- Up-to-date: /usr/local/include/jetson-utils/gstEncoder.h
-- Up-to-date: /usr/local/include/jetson-utils/gstUtility.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaBayer.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaColormap.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaColorspace.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaCrop.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaFilterMode.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaFont.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaGrayscale.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaMappedMemory.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaMath.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaNormalize.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaOverlay.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaPointCloud.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaRGB.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaResize.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaUtility.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaVector.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaWarp.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaYUV.h
-- Up-to-date: /usr/local/include/jetson-utils/cudaAlphaBlend.cuh
-- Up-to-date: /usr/local/include/jetson-utils/cudaFilterMode.cuh
-- Up-to-date: /usr/local/include/jetson-utils/glBuffer.h
-- Up-to-date: /usr/local/include/jetson-utils/glCamera.h
-- Up-to-date: /usr/local/include/jetson-utils/glDisplay.h
-- Up-to-date: /usr/local/include/jetson-utils/glEvents.h
-- Up-to-date: /usr/local/include/jetson-utils/glTexture.h
-- Up-to-date: /usr/local/include/jetson-utils/glUtility.h
-- Up-to-date: /usr/local/include/jetson-utils/glWidget.h
-- Up-to-date: /usr/local/include/jetson-utils/imageFormat.h
-- Up-to-date: /usr/local/include/jetson-utils/imageIO.h
-- Up-to-date: /usr/local/include/jetson-utils/imageLoader.h
-- Up-to-date: /usr/local/include/jetson-utils/imageWriter.h
-- Up-to-date: /usr/local/include/jetson-utils/loadImage.h
-- Up-to-date: /usr/local/include/jetson-utils/imageFormat.inl
-- Up-to-date: /usr/local/include/jetson-utils/devInput.h
-- Up-to-date: /usr/local/include/jetson-utils/devJoystick.h
-- Up-to-date: /usr/local/include/jetson-utils/devKeyboard.h
-- Up-to-date: /usr/local/include/jetson-utils/Endian.h
-- Up-to-date: /usr/local/include/jetson-utils/IPv4.h
-- Up-to-date: /usr/local/include/jetson-utils/NetworkAdapter.h
-- Up-to-date: /usr/local/include/jetson-utils/Socket.h
-- Up-to-date: /usr/local/include/jetson-utils/Event.h
-- Up-to-date: /usr/local/include/jetson-utils/Mutex.h
-- Up-to-date: /usr/local/include/jetson-utils/Process.h
-- Up-to-date: /usr/local/include/jetson-utils/RingBuffer.h
-- Up-to-date: /usr/local/include/jetson-utils/Thread.h
-- Up-to-date: /usr/local/include/jetson-utils/Event.inl
-- Up-to-date: /usr/local/include/jetson-utils/Mutex.inl
-- Up-to-date: /usr/local/include/jetson-utils/RingBuffer.inl
-- Up-to-date: /usr/local/include/jetson-utils/videoOptions.h
-- Up-to-date: /usr/local/include/jetson-utils/videoOutput.h
-- Up-to-date: /usr/local/include/jetson-utils/videoSource.h
-- Up-to-date: /usr/local/lib/libjetson-utils.so
-- Up-to-date: /usr/local/share/jetson-utils/cmake/jetson-utilsConfig.cmake
-- Up-to-date: /usr/local/share/jetson-utils/cmake/jetson-utilsConfig-noconfig.cmake
-- Up-to-date: /usr/local/bin/camera-viewer
-- Up-to-date: /usr/local/bin/video-viewer
-- Up-to-date: /usr/local/bin/gl-display-test
-- Up-to-date: /usr/local/bin/camera-viewer.py
-- Up-to-date: /usr/local/bin/cuda-examples.py
-- Up-to-date: /usr/local/bin/cuda-from-cv.py
-- Up-to-date: /usr/local/bin/cuda-from-numpy.py
-- Up-to-date: /usr/local/bin/cuda-to-cv.py
-- Up-to-date: /usr/local/bin/cuda-to-numpy.py
-- Up-to-date: /usr/local/bin/gl-display-test.py
-- Up-to-date: /usr/local/bin/video-viewer.py
-- Up-to-date: /usr/lib/python2.7/dist-packages/jetson_utils_python.so
-- Up-to-date: /usr/lib/python2.7/dist-packages/Jetson
-- Up-to-date: /usr/lib/python2.7/dist-packages/Jetson/Utils
-- Up-to-date: /usr/lib/python2.7/dist-packages/Jetson/Utils/__init__.py
-- Installing: /usr/lib/python2.7/dist-packages/Jetson/__init__.py
-- Up-to-date: /usr/lib/python2.7/dist-packages/jetson
-- Up-to-date: /usr/lib/python2.7/dist-packages/jetson/utils
-- Up-to-date: /usr/lib/python2.7/dist-packages/jetson/utils/__init__.py
-- Up-to-date: /usr/lib/python2.7/dist-packages/jetson/__init__.py
-- Up-to-date: /usr/lib/python3.6/dist-packages/jetson_utils_python.so
-- Up-to-date: /usr/lib/python3.6/dist-packages/Jetson
-- Up-to-date: /usr/lib/python3.6/dist-packages/Jetson/Utils
-- Up-to-date: /usr/lib/python3.6/dist-packages/Jetson/Utils/__init__.py
-- Up-to-date: /usr/lib/python3.6/dist-packages/Jetson/__init__.py
-- Up-to-date: /usr/lib/python3.6/dist-packages/jetson
-- Up-to-date: /usr/lib/python3.6/dist-packages/jetson/utils
-- Up-to-date: /usr/lib/python3.6/dist-packages/jetson/utils/__init__.py
-- Up-to-date: /usr/lib/python3.6/dist-packages/jetson/__init__.py
-- Up-to-date: /usr/local/bin/detectnet.py
-- Up-to-date: /usr/local/bin/imagenet.py
-- Up-to-date: /usr/local/bin/my-detection.py
-- Up-to-date: /usr/local/bin/my-recognition.py
-- Up-to-date: /usr/local/bin/segnet.py
-- Up-to-date: /usr/local/bin/segnet_utils.py
-- Up-to-date: /usr/local/bin/imagenet-console.py
-- Up-to-date: /usr/local/bin/imagenet-camera.py
-- Up-to-date: /usr/local/bin/detectnet-console.py
-- Up-to-date: /usr/local/bin/detectnet-camera.py
-- Up-to-date: /usr/local/bin/segnet-console.py
-- Up-to-date: /usr/local/bin/segnet-camera.py
-- Up-to-date: /usr/lib/python2.7/dist-packages/jetson_inference_python.so
-- Up-to-date: /usr/lib/python2.7/dist-packages/Jetson
-- Up-to-date: /usr/lib/python2.7/dist-packages/Jetson/Inference
-- Up-to-date: /usr/lib/python2.7/dist-packages/Jetson/Inference/__init__.py
-- Installing: /usr/lib/python2.7/dist-packages/Jetson/__init__.py
CMake Error at python/bindings_python_2.7/cmake_install.cmake:89 (file):
  file INSTALL cannot find
  "/home/dlinano/Downloads/jetson-inference/python/bindings/../python/jetson".
Call Stack (most recent call first):
  python/cmake_install.cmake:90 (include)
  cmake_install.cmake:121 (include)

Makefile:117: recipe for target 'install' failed
make: *** [install] Error 1

lc150303, 2020-09-01:
have you figured out solution?

yangs12, 2020-09-02:
not yet...

lc150303, 2020-09-02:
> not yet...

It's case error of the dir name. **It's named 'Jetson' not 'jetson'.**

Change the "/home/dlinano/Downloads/jetson-inference/python/bindings/../python/**jetson**" into "/home/dlinano/Downloads/jetson-inference/python/bindings/../python/**Jetson**" in the cmake files mentioned in the error message, which are in "(your build dir)/utils/python/bindings_xxx/" or "(your build dir)/python/bindings_xxx/".

This works for me.

yangs12, 2020-09-02:
Thank you so much. I will try this.

Maxcharm, 2020-11-02:
Have you solved this problem? Because I encountered an error which is exactly the same.

feiwei93, 2021-01-08:
Thank you so much. This works for me.

ihongxx, 2021-07-06:
> 这是目录名称的大小写错误。**它被命名为“Jetson”而不是“jetson”。**
> 
> 更改“/home/dlinano/Downloads/jetson-inference/python/bindings/../python/**杰特森**”到“/home/dlinano/Downloads/jetson-inference/python/bindings/../python/**特森**”在错误消息中提到的 cmake 文件，它们位于“（您的构建目录）/utils/python/bindings_xxx/”或“（您的构建目录）/python/bindings_xxx/”中。
> 
> 这对我有用。

ihongxx, 2021-07-06:
> It's case error of the dir name. **It's named 'Jetson' not 'jetson'.**
>
> Change the "/home/dlinano/Downloads/jetson-inference/python/bindings/../python/**jetson**" into [...]

Thank you so much. This works for me.

Bian-666, 2022-05-12:
The problem is your 'jetson-inference/python/python' only have 'Jetson'folder, not have 'jetson' folder. 
To download 'jetson-inference.zip' in github, then copy the 'jetson-inference/python/python/jeston' folder into your 'jetson-inference/python/python'. 

This works for me.

closed 2023-04-27 by dusty-nv