dusty-nv/jetson-inference #684: 'jetson.utils' has no attribute 'videoSource'

JVTEAM, 2020-08-18:
I'm attempting to run my-detection.py using an rtsp feed on my network.

Using the instructions it said I can add it modifying this line.

#camera = jetson.utils.videoSource("rtsp://*****")

However, jetson.utils is reporting this...

AttributeError: module 'jetson.utils' has no attribute 'videoSource'
PyTensorNet_Dealloc()

dusty-nv, 2020-08-18:
@JVTEAM do the other python samples run that also use videoSource, or is it just after you perform this modification?

Can you run with RTSP from the command line?   i.e.

```bash
$ python3 video-viewer.py rtsp://<IP>:<PORT>
```

JVTEAM, 2020-08-18:
same issue..

~/jetson-inference/python/examples$ python3 video-viewer.py rtsp://office:Lessee77@192.168.86.56/live
jetson.utils.__init__.py
jetson.utils -- initializing Python 3.6 bindings...
jetson.utils -- registering module functions...
jetson.utils -- done registering module functions
jetson.utils -- registering module types...
jetson.utils -- done registering module types
jetson.utils -- done Python 3.6 binding initialization
Traceback (most recent call last):
  File "video-viewer.py", line 33, in <module>
    epilog=jetson.utils.videoSource.Usage() + jetson.utils.videoOutput.Usage() + jetson.utils.logUsage())
AttributeError: module 'jetson.utils' has no attribute 'videoSource'

dusty-nv, 2020-08-18:
It seems like your jetson-utils may be on an older version, because on the updates that supported videoSource, this debug text was hidden:

```
jetson.utils -- initializing Python 3.6 bindings...
jetson.utils -- registering module functions...
jetson.utils -- done registering module functions
jetson.utils -- registering module types...
jetson.utils -- done registering module types
jetson.utils -- done Python 3.6 binding initialization
```

So you would no longer see that if you were on the latest.

Can you try running `sudo make install` from your `jetson-inference/build` directory?

If that doesn't work, please re-clone the jetson-inference master and re-build/re-install it.

JVTEAM, 2020-08-18:
> It seems like your jetson-utils may be on an older version, because on the updates that supported videoSource, this debug text was hidden:
>
> ```
> jetson.utils -- initializing Python 3.6 bindings... [...]

I just freshly re-built it yesterday, so I don't think that is the issue.

JVTEAM, 2020-08-18:
I know I rebuilt it on my tx2, just rebuilding on the nano just to make sure.  Back in a bit..

dusty-nv, 2020-08-18:
> I just freshly re-built it yesterday, so I don't think that is the issue.

Just double-checked, and that debug text is no longer produced on master.   So it seems like when you updated/re-built, the new Python modules weren't installed.   Did you run `sudo make install` (this is what installs the Python modules)?

Perhaps you want to clean out the jetson-inference/jetson-utils modules under `/usr/lib/python3.6/dist-packages` to be sure

JVTEAM, 2020-08-18:
Yeah, I think I got distracted yesterday when I built it on the tx2.  I accidentally clicked install on that pytorch option in the installer.  That installs a really old non-4.4 friendly version that I had to fix.  

looks like we're good on the first issue.  Now a new one.

[gstreamer] gstDecoder -- creating decoder for rtsp://user:pass@IP:1234
[gstreamer] gstDecoder -- Could not open resource for reading and writing.
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://user:pass@IP:1234
Traceback (most recent call last):
  File "my-detection.py", line 30, in <module>
    camera = jetson.utils.videoSource("rtsp://user:pass@IP:1234")
Exception: jetson.utils -- failed to create videoSource device

dusty-nv, 2020-08-18:
It's unable to discover the attributes of your RTSP stream.   Try adding `--input-codec=h264` (or whatever the correct codec is) to the command line options:

```
$ python3 video-viewer.py --input-codec=h264 rtsp://user:pass@IP:1234
```

I recommend trying the video-viewer first just to make sure you can get the feed before proceeding.

JVTEAM, 2020-08-18:
Ok, it works in the video viewer script.  I'll try and mine that code until it decides to work in the other.

closed 2020-08-18 by JVTEAM

Samuel-wei, 2020-10-11:
> It's unable to discover the attributes of your RTSP stream. Try adding `--input-codec=h264` (or whatever the correct codec is) to the command line options:
>
> ```
> $ python3 video-viewer.py [...]

I try using above command line got the error:
$ python3 video-viewer.py --input-codec=h264 rtsp://admin:123456@192.168.0.200:554
output error:
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for admin:123456@192.168.0.200
[gstreamer] gstDecoder -- Could not open resource for reading.
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://admin:123456@192.168.0.200:554 ! queue ! rtph264depay ! h264parse ! omxh264dec ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from rtsp://admin:123456@192.168.0.200:554
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: rtsp://admin:123456@192.168.0.200:554
     - protocol:  rtsp
     - location:  admin:123456@192.168.0.200
     - port:      554
  -- deviceType: ip
  -- ioType:     input
  -- codec:      h264
  -- width:      0
  -- height:     0
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] failed to open X11 server connection.
[OpenGL] failed to create X11 Window.
video-viewer:  failed to create output stream
[gstreamer] opening gstDecoder for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from NULL to READY ==> h264parse0
[gstreamer] gstreamer changed state from NULL to READY ==> rtph264depay0
[gstreamer] gstreamer changed state from NULL to READY ==> queue0
[gstreamer] gstreamer changed state from NULL to READY ==> rtspsrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer changed state from READY to PAUSED ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from READY to PAUSED ==> h264parse0
[gstreamer] gstreamer changed state from READY to PAUSED ==> rtph264depay0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> queue0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> rtspsrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> h264parse0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> rtph264depay0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> queue0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> rtspsrc0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer rtspsrc0 ERROR Could not open resource for reading.
[gstreamer] gstreamer Debugging info: gstrtspsrc.c(5829): gst_rtspsrc_setup_auth (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
No supported authentication protocol was found
[gstreamer] gstreamer rtspsrc0 ERROR Not found
[gstreamer] gstreamer Debugging info: gstrtspsrc.c(6112): gst_rtspsrc_send (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Not Found (404)
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer message progress ==> rtspsrc0
video-viewer:  failed to capture video frame
..........

what's wrong of my operating?

Samuel-wei, 2020-10-11:
> It's unable to discover the attributes of your RTSP stream. Try adding `--input-codec=h264` (or whatever the correct codec is) to the command line options:
>
> ```
> $ python3 video-viewer.py [...]

my host is jetson nano
v4l2 csi camera was working
I use the above command line got error:
`[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for admin:123456@192.168.0.200
[gstreamer] gstDecoder -- Could not open resource for reading.
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://admin:123456@192.168.0.200:554 ! queue ! rtph264depay ! h264parse ! omxh264dec ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from rtsp://admin:123456@192.168.0.200:554
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: rtsp://admin:123456@192.168.0.200:554
     - protocol:  rtsp
     - location:  admin:123456@192.168.0.200
     - port:      554
  -- deviceType: ip
  -- ioType:     input
  -- codec:      h264
  -- width:      0
  -- height:     0
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] failed to open X11 server connection.
[OpenGL] failed to create X11 Window.
video-viewer:  failed to create output stream
[gstreamer] opening gstDecoder for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from NULL to READY ==> h264parse0
[gstreamer] gstreamer changed state from NULL to READY ==> rtph264depay0
[gstreamer] gstreamer changed state from NULL to READY ==> queue0
[gstreamer] gstreamer changed state from NULL to READY ==> rtspsrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer changed state from READY to PAUSED ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from READY to PAUSED ==> h264parse0
[gstreamer] gstreamer changed state from READY to PAUSED ==> rtph264depay0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> queue0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> rtspsrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> h264parse0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> rtph264depay0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> queue0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> rtspsrc0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer rtspsrc0 ERROR Could not open resource for reading.
[gstreamer] gstreamer Debugging info: gstrtspsrc.c(5829): gst_rtspsrc_setup_auth (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
No supported authentication protocol was found
[gstreamer] gstreamer rtspsrc0 ERROR Not found
[gstreamer] gstreamer Debugging info: gstrtspsrc.c(6112): gst_rtspsrc_send (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Not Found (404)
[gstreamer] gstreamer message progress ==> rtspsrc0
[gstreamer] gstreamer message progress ==> rtspsrc0
video-viewer:  failed to capture video frame
..................`
What's wrong of my operational?

landonpx, 2022-11-09:
> @JVTEAM do the other python samples run that also use videoSource, or is it just after you perform this modification?
> 
> Can you run with RTSP from the command line? i.e.
> 
> ```shell
> $ python3 video-viewer.py rtsp://<IP>:<PORT>
> ```

Hi,
I freshly built jetson-inference and jetson-utils, when I try to run video-viewer.py in all methods mentioned above I get:
ImportError: cannot import name 'videoSource'
Please help.
Thanks

dusty-nv, 2022-11-09:
Hi @landonpx, did you run `sudo make install && sudo ldconfig` ?

landonpx, 2022-11-09:
> Hi @landonpx, did you run `sudo make install && sudo ldconfig` ?

Yes

dusty-nv, 2022-11-10:
Do you happen to be using a virtualenv, and does this also happen when using the container?  ([`docker/run.sh`](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md))

landonpx, 2022-11-10:
> Do you happen to be using a virtualenv, and does this also happen when using the container? ([`docker/run.sh`](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md))

Not using virtualenv, 
I tried using the container and I am able to run video-viewer