dusty-nv/jetson-inference #685: Can't get my-detection (modified) to display output

JVTEAM, 2020-08-18:
I used the code from here:
https://raw.githubusercontent.com/dusty-nv/jetson-utils/master/python/examples/video-viewer.py

to modify my-detection as the video viewer code worked.  However, it's slightly different in the output area.

Here's what I did:

import jetson.inference
import jetson.utils
import argparse
import sys

# parse command line
parser = argparse.ArgumentParser(description="View various types of video streams", 
                                 formatter_class=argparse.RawTextHelpFormatter, 
                                 epilog=jetson.utils.videoSource.Usage() + jetson.utils.videoOutput.Usage() + jetson.utils.logUsage())

parser.add_argument("input_URI", type=str, help="URI of the input stream")
parser.add_argument("output_URI", type=str, default="", nargs='?', help="URI of the output stream")
try:
	opt = parser.parse_known_args()[0]
except:
	print("")
	parser.print_help()
	sys.exit(0)
net = jetson.inference.detectNet("pednet", threshold=0.5)
input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)
output = jetson.utils.videoOutput(opt.output_URI, argv=sys.argv)

while output.IsStreaming():
	img = input.Capture()
	detections = net.Detect(img)
	output.Render(img)
	#output.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
	output.SetStatus("Video Viewer | {:d}x{:d} | {:.1f} FPS".format(img.width, img.height, output.GetFrameRate()))

I then put in:
python3 my-detection.py --input-codec=h264 rtsp://user:pass@ip/live
into the CLI

This works with the video-viewer.  But once I put in the detection stuff, it just shows this:

[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)150000, maximum-bitrate=(uint)11745960, bitrate=(uint)823347;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)150000, maximum-bitrate=(uint)11745960, bitrate=(uint)803285;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)150000, maximum-bitrate=(uint)11745960, bitrate=(uint)826347;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", minimum-bitrate=(uint)150000, maximum-bitrate=(uint)11745960, bitrate=(uint)808162;

So it appears to be working, but not displaying.

JVTEAM, 2020-08-18:
[video]  created gstDecoder from rtsp://*****
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: rtsp://*****
     - protocol:  rtsp
     - location:  office
     - port:      554
  -- deviceType: ip
  -- ioType:     input
  -- codec:      h264
  -- width:      1920
  -- height:     1080
  -- frameRate:  15.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[OpenGL] failed to create X11 Window.
jetson.utils -- no output streams, creating fake null output

Failed to create an x11 window...

dusty-nv, 2020-08-18:
Does it work if you use `detectnet.py` ?

You might want to try creating `output = jetson.utils.videoOutput(opt.output_URI, argv=sys.argv)` before the detection network to see if that changes anything.

JVTEAM, 2020-08-18:
Same error in detectnet.py

JVTEAM, 2020-08-18:
i simplified it to the exact same result to be more like the custom my-detection.py

import jetson.inference
import jetson.utils

camera = jetson.utils.videoSource("rtsp://****")
display = jetson.utils.videoOutput("display://0") # 'my_video.mp4' for file
net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)

while display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	display.Render(img)
	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))

JVTEAM, 2020-08-18:
I've seen a bunch of posts, but I don't see resolution to them about the x11 server issue.  Both the computers I'm testing have VNC running on them.

dusty-nv, 2020-08-18:
Hmm, that is strange.   Does this work for you?

```python
import jetson.utils

camera = jetson.utils.videoSource("rtsp://****")
display = jetson.utils.videoOutput("display://0") # 'my_video.mp4' for file

import jetson.inference
net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)

while display.IsStreaming():
   img = camera.Capture()
   detections = net.Detect(img)
   display.Render(img)
   display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
```

JVTEAM, 2020-08-18:
Yes, it does!  perhaps update the github page with the new order.

closed 2020-08-18 by JVTEAM

dusty-nv, 2020-08-18:
Unfortunately I haven't been able to reproduce it on any Jetson here, but I will try to look more into why it happened for you.

Jacquesvw, 2020-10-10:
> Hmm, that is strange. Does this work for you?
>
> ```python
> import jetson.utils
>
> camera = jetson.utils.videoSource("rtsp://****")
> display = jetson.utils.videoOutput("display://0") # 'my_video.mp4' for [...]

Thanks, solved the same issue for me.

kopterkraft, 2020-12-27:
I can confirm that swapping the import order and defining the net after setting input and output solved my issue when reading from a rtsp stream too. Before that, I also got not output on display 0.
(and I also have VNC running, as it was also mentioned by JVTEAM)