dusty-nv/jetson-inference #687: How to classify just one class in segnet?

Daniel-10, 2020-08-18:
Hey Dusty,   
I'd like to get the pixels of cars and then make my own mask for them from another image.   I'm just a hobbyist but do have a couple of questions related to this.

- Should I just make it ignore all other classes except car class?  (It seems like the probability would be above some threshold, maybe .5 or something if it matches a car?)  
- Where would the class list IDs be?  
- Would it be ok to do the image pixel swap on the cpu using C instead of cuda? (I know very, very little about how cuda works)
- For these networks is Tensornet.cpp,  is the output format common to TensorRT?   ( I dont think i need to know anything about it yet but am just curious ).  Eventually I'd like to run pose-trt alongside this one.

Thanks, Daniel

I saw this code in segnet.cpp

`	uint8_t* classMap = mClassMap[0];

	for( uint32_t y=0; y < s_h; y++ )
	{
		for( uint32_t x=0; x < s_w; x++ )
		{
			float p_max = -100000.0f;
			int   c_max = -1;

			for( int c=0; c < s_c; c++ )
			{
				// skip ignoreID
				if( c == ignoreID )
					continue;

				// check if this class score is higher
				const float p = scores[c * s_w * s_h + y * s_w + x];

				if( c_max < 0 || p > p_max )
				{
					p_max = p;
					c_max = c;
				}
			}

			classMap[y * s_w + x] = c_max;
			//printf("(%u, %u) -> class %i\n", x, y, (uint32_t)c_max);
		}
	}
`
It seems to be going through each pixel and checking the probabilities of each class like they talk about in the segnet paper

> The output of the soft-max
> classifier is a K channel image of probabilities where K is the
> number of classes. The predicted segmentation corresponds to the
> class with maximum probability at each pixel.

dusty-nv, 2020-08-18:
Hi @Daniel-10 , are you using Python or C++?

You can get that class mask from [`segNet::Mask()`](https://github.com/dusty-nv/jetson-inference/blob/4e6b7ff37935a9bc64271119f42cd25366ce8c79/c/segNet.h#L256) function (there is a similar way to access it from Python) and then look for the classID that corresponds to car.

You can find the class ID with [`segNet::FindClassID()`](https://github.com/dusty-nv/jetson-inference/blob/4e6b7ff37935a9bc64271119f42cd25366ce8c79/c/segNet.h#L296)

Daniel-10, 2020-08-18:
@dusty-nv,  Either Python or C++ would be ok.  If it's possible I'd like to start with python so I dont have to compile while I'm trying stuff out.  

I think I understand what to do from your explanation.  What would be the python calls to access the class mask and classID?   

Plus thanks a lot for making these tutorials so accessible and for giving the guidance!  It's very exciting to get these things running!

Daniel-10, 2020-08-18:
@dusty-nv,  are these the two python functions to use?

`	{ "Mask", (PyCFunction)PySegNet_Mask, METH_VARARGS|METH_KEYWORDS, DOC_MASK},	
	{ "GetClassDesc", (PyCFunction)PySegNet_GetClassDesc, METH_VARARGS, DOC_GET_CLASS_DESC},`

Daniel-10, 2020-08-18:
I think I'm good to go.  Thanks again for the guidance!

closed 2020-08-18 by Daniel-10

dusty-nv, 2020-08-18:
Yep you got it, pass a `gray8` image to `segNet.Mask()` and it will give you the class ID's back.

It is how the `--stats` option is implemented in the `segnet.py` example:

https://github.com/dusty-nv/jetson-inference/blob/4e6b7ff37935a9bc64271119f42cd25366ce8c79/python/examples/segnet_utils.py#L72

You can either get the class ID's at the raw resolution, which is `grid_width, grid_height = net.GetGridSize()`
That will be much smaller than the original resolution of the input image, so the processing overhead is less.

However if you want it upscaled, you can just pass in a larger buffer and it will upscale the classID's for you.

Daniel-10, 2020-08-20:
@dusty-nv ,  wow, that was much easier than i expected.  It seems pretty straightforward to get trt_pose in there as well.  I'm glad yall have made these fairly easy to handle.  Thank you for the hobbyist support.  I cant wait to show the kids during their social distant car parade they can make their car disappear by waving their hands around.