dusty-nv/jetson-inference #688: ONNX Export Issue (error in loading state_dict for SSD)

izmevid, 2020-08-19:
I would be very grateful about your help:

I am trying to export my trained model which is based on mobile net-v1-ssd-mp-0_675.pth to onnx. 

When I want to do this as described in "https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md", I am getting the following error:

[image]

Your help is very much appreciated @dusty-nv !!!

izmevid, 2020-08-19:
could this be because of the jetpack version 4.3?

Version: 4.3-b134 is installed..

drmnasr, 2020-10-06:
Hello,
in case you are still facing this or anyone came across this problem, reference your labels file correctly and make sure it has "BACKGROUND" as one of the labels

michael-herm, 2020-11-19:
Hello,
Im facing the same issue.
I trained my model based on the "Collecting your own Detection Dataset".
It looks like the training went ssucessfully.
In the "/ssd/models/sweets3" directory theres the new "labels.txt" file with the  added "BACKGROUND" class and the "mb1-ssd-Epoch-0-Loss-3.47.pth" file, but NOT the "ssd-mobilenet.onnx" model.
@dusty-nv do you know what could cause this problem?

root@michael-desktop:/jetson-inference/python/training/detection/ssd# python3 train_ssd.py --dataset-type=voc --data=data/sweets3 --model-dir=models/sweets3 --batch-size=2 --workers=1 --epochs=1
2020-11-19 08:21:36 - Using CUDA...
2020-11-19 08:21:36 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=2, checkpoint_folder='models/sweets3', dataset_type='voc', datasets=['data/sweets3'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=1, num_workers=1, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2020-11-19 08:21:36 - Prepare training datasets.
2020-11-19 08:21:36 - VOC Labels read from file: ('BACKGROUND', 'airwaves', 'icebreakers', '')
2020-11-19 08:21:36 - Stored labels into file models/sweets3/labels.txt.
2020-11-19 08:21:36 - Train dataset size: 182
2020-11-19 08:21:36 - Prepare Validation datasets.
2020-11-19 08:21:36 - VOC Labels read from file: ('BACKGROUND', 'airwaves', 'icebreakers', '')
2020-11-19 08:21:36 - Validation dataset size: 20
2020-11-19 08:21:36 - Build network.
2020-11-19 08:21:36 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
2020-11-19 08:21:37 - Took 0.50 seconds to load the model.
2020-11-19 08:21:43 - Learning rate: 0.01, Base net learning rate: 0.001, Extra Layers learning rate: 0.01.
2020-11-19 08:21:43 - Uses CosineAnnealingLR scheduler.
2020-11-19 08:21:43 - Start training from epoch 0.
/usr/local/lib/python3.6/dist-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
/usr/local/lib/python3.6/dist-packages/torch/nn/_reduction.py:44: UserWarning: size_average and reduce args will be deprecated, please use reduction='sum' instead.
  warnings.warn(warning.format(ret))
2020-11-19 08:22:24 - Epoch: 0, Step: 10/91, Avg Loss: 12.5837, Avg Regression Loss 4.2654, Avg Classification Loss: 8.3182
2020-11-19 08:22:33 - Epoch: 0, Step: 20/91, Avg Loss: 7.8180, Avg Regression Loss 3.0547, Avg Classification Loss: 4.7633
2020-11-19 08:22:38 - Epoch: 0, Step: 30/91, Avg Loss: 6.3763, Avg Regression Loss 2.4139, Avg Classification Loss: 3.9624
2020-11-19 08:22:43 - Epoch: 0, Step: 40/91, Avg Loss: 5.5971, Avg Regression Loss 2.0275, Avg Classification Loss: 3.5696
2020-11-19 08:22:48 - Epoch: 0, Step: 50/91, Avg Loss: 5.9933, Avg Regression Loss 2.3471, Avg Classification Loss: 3.6463
2020-11-19 08:23:01 - Epoch: 0, Step: 60/91, Avg Loss: 5.4874, Avg Regression Loss 2.1665, Avg Classification Loss: 3.3209
2020-11-19 08:23:12 - Epoch: 0, Step: 70/91, Avg Loss: 5.3203, Avg Regression Loss 2.1231, Avg Classification Loss: 3.1972
2020-11-19 08:23:18 - Epoch: 0, Step: 80/91, Avg Loss: 4.9167, Avg Regression Loss 1.8312, Avg Classification Loss: 3.0855
2020-11-19 08:23:28 - Epoch: 0, Step: 90/91, Avg Loss: 5.2025, Avg Regression Loss 2.3432, Avg Classification Loss: 2.8593
2020-11-19 08:23:31 - Epoch: 0, Validation Loss: 3.4752, Validation Regression Loss 1.1896, Validation Classification Loss: 2.2856
2020-11-19 08:23:31 - Saved model models/sweets3/mb1-ssd-Epoch-0-Loss-3.475229287147522.pth
2020-11-19 08:23:31 - Task done, exiting program.
Segmentation fault (core dumped)
root@michael-desktop:/jetson-inference/python/training/detection/ssd# python3 onnx_export.py --model-dir=models/sweets3
Namespace(batch_size=1, height=300, input='', labels='labels.txt', model_dir='models/sweets3', net='ssd-mobilenet', output='', width=300)
running on device cuda:0
found best checkpoint with loss 3.475229 (models/sweets3/mb1-ssd-Epoch-0-Loss-3.475229287147522.pth)
creating network:  ssd-mobilenet
num classes:       3
loading checkpoint:  models/sweets3/mb1-ssd-Epoch-0-Loss-3.475229287147522.pth
Traceback (most recent call last):
  File "onnx_export.py", line 86, in <module>
    net.load(args.input)
  File "/jetson-inference/python/training/detection/ssd/vision/ssd/ssd.py", line 135, in load
    self.load_state_dict(torch.load(model, map_location=lambda storage, loc: storage))
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 1045, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for SSD:
	size mismatch for classification_headers.0.weight: copying a param with shape torch.Size([24, 512, 3, 3]) from checkpoint, the shape in current model is torch.Size([18, 512, 3, 3]).
[... cut ...]
	size mismatch for classification_headers.4.bias: copying a param with shape torch.Size([24]) from checkpoint, the shape in current model is torch.Size([18]).
	size mismatch for classification_headers.5.weight: copying a param with shape torch.Size([24, 256, 3, 3]) from checkpoint, the shape in current model is torch.Size([18, 256, 3, 3]).
	size mismatch for classification_headers.5.bias: copying a param with shape torch.Size([24]) from checkpoint, the shape in current model is torch.Size([18]).
root@michael-desktop:/jetson-inference/python/training/detection/ssd#

dusty-nv, 2020-11-19:
> VOC Labels read from file: ('BACKGROUND', 'airwaves', 'icebreakers', '')

Hi @michael-herm , is it possible that your labels.txt has an empty line at the end?    

If so, try removing the empty line from your original labels.txt, then train the model again for an epoch, then try to export to ONNX again.

michael-herm, 2020-11-19:
That was it.
Thanks a lot! @dusty-nv

closed 2023-04-27 by dusty-nv