dusty-nv/jetson-inference #689: Issue with camera (flip method)

spy-maker, 2020-08-21:
I am working on this using jetson nano. The current xavier version camera code is making my output display upside down because of flip method. May I know how to modify the codes to display the camera normally.

dusty-nv, 2020-08-21:
If you are on master, you can use the --flip-method command line option.  Run the program with --help to see the options to it.

spy-maker, 2020-08-21:
Do I need to change this part in gstCamera.cpp file?[image]

spy-maker, 2020-08-25:
> If you are on master, you can use the --flip-method command line option.  Run the program with --help to see the options to it.
> From: spy-maker <notifications@github.com>
> Sent: Thursday, August 20, [...]

Hi,
I still cannot fix the issue. I tried to modify the codes in gstcamera.cpp file but I still can't solve it.
Can you provide me a detailed step on how to fix the flip method issue?

spy-maker, 2020-08-25:
Hi,
I still cannot fix the issue. I tried to modify the codes in gstcamera.cpp file but I still can't solve it.
Can you provide me a detailed step on how to fix the flip method issue?

dusty-nv, 2020-08-25:
If you update your repo to master, you don't need to edit any file anymore.  You can just use the `--flip-method` command line argument:

```bash
$ video-viewer.py --flip-method=rotate-180 csi://0
```

spy-maker, 2020-08-26:
> If you update your repo to master, you don't need to edit any file anymore.  You can just use the `--flip-method` command line argument:
> 
> ```bash
> $ video-viewer.py --flip-method=rotate-180 csi://0
> ```

Thanks for reply. Now I can the see the normal display. But how can I make this change applicable to all the programs? 
Currently only the camera output for video-viewer.py is solved.

spy-maker, 2020-08-26:
> If you update your repo to master, you don't need to edit any file anymore.  You can just use the `--flip-method` command line argument:
> 
> ```bash
> $ video-viewer.py --flip-method=rotate-180 csi://0
> ```

Thanks for the reply. The problem is fixed for few program files. Is there any steps to make this change permanent instead of typing everytime in Terminal?

mfreyne, 2020-09-16:
Hi, I'm looking for the same answer as spy-maker

dusty-nv, 2020-09-16:
@mfreyne , have you tried running with `--flip-method=rotate-180` argument?

mfreyne, 2020-09-17:
Hi dusty-nv,
I have not tried this as I'm unsure where to put this code, can I place it in my-detection.py?
or
in gstCamera.cpp?
Hints and help greatly appreciated!

Thank you

dusty-nv, 2020-09-17:
`--flip-method=rotate-180` is a command line argument to the detectnet/imagenet/segnet/video-viewer sample programs.

If you wanted to add support for this argument to `my-detection.py`, change the videoSource line to:

```python
import sys
camera = jetson.utils.videoSource("csi://0", argv=sys.argv)
```

mfreyne, 2020-09-17:
Sorry dusty-nv
I think that I am in over my depth, I cant figure it out

mfreyne, 2020-09-17:
dusty-nv

The penny just dropped, I got it working
Thank you so much and I really enjoy this Jetson Nano

dusty-nv, 2020-09-17:
OK cool, glad you got it working @mfreyne !

closed 2021-03-15 by spy-maker

peter58-blip, 2021-04-14:
i am implementing this on the TX2 i am having the same issue of inverted image. i tried the suggested comments above and changing the code but does not seem to work. Could anyone help me out?

dusty-nv, 2021-04-14:
Hi @peter58-blip, does running `video-viewer` utility with `--flip-method=rotate-180` result in the correct orientation of the image?

AndresRodriguezTinoco, 2021-06-30:
https://gstreamer.freedesktop.org/documentation/videofilter/videoflip.html?gi-language=c
`video-viewer` `--flip-method=`
none (0) – Identity (no rotation)
`clockwise` (1) – Rotate clockwise 90 degrees
`rotate-180` (2) – Rotate 180 degrees
`counterclockwise` (3) – Rotate counter-clockwise 90 degrees
`horizontal-flip` (4) – Flip horizontally
`vertical-flip` (5) – Flip vertically
`upper-left-diagonal` (6) – Flip across upper left/lower right diagonal
`upper-right-diagonal` (7) – Flip across upper right/lower left diagonal
`automatic` (8) – Select flip method based on image-orientation tag

captain-crypto, 2022-02-02:
The "--flip-method" is not applicable for webcam (from the docs :" flip method to apply to input (excludes V4L2)".
is there a way to to flip with image on the GPU (without opencv)?

dusty-nv, 2022-02-02:
Hi @captain-crypto, I just added support for `--input-flip` method for V4L2 webcams in https://github.com/dusty-nv/jetson-inference/commit/9c255b9e48229665428e086f2e268d21c7763314

Can you let me know if it works for you?

Giacomo76, 2022-09-09:
I am currently having problems with the FLIP method with picamera csi: // 0
the video stream of the camera is upside down.
When I try with ./detectnet.py --flip-method = horizontal csi: // 0 ---> flip vertical

./detectnet.py --flip-method = vertical csi: // 0 ---> flip horizontal
./detectnet.py --flip-method = rotate-180 csi: // 0 ---> crash or freeze why ??????
./detectnet.py --flip-method = clockwise csi: // 0 ---> counterclockwise
./detectnet.py --flip-method = counterclockwise csi: // 0 ---> clockwise

I have the same problem using --input-flip
./detectnet.py --input-flip = vertical csi: // 0 ---> flip horizontal
./detectnet.py --input-flip = rotate-180 csi: // 0 ---> crash or freeze why ??????
./detectnet.py --input-flip = clockwise csi: // 0 ---> counterclockwise
./detectnet.py --input-flip = counterclockwise csi: // 0 ---> clockwise
What can this malfunction depend on?I leave a photo of my jetson nano environment

[image]

Giacomo76, 2022-09-09:
For example:
jack@jack-desktop:~/jetson-inference/build/aarch64/bin$ ./detectnet.py --flip-method=rotate-180  --network=ssd-inception-v2
[image]

raulo987, 2023-04-30:
Same problem here. :)

While using --inpute-rate=rotate-180 in python code and also while trying to change flip method in gstcamera.cpp. 

In python code: 

camera1 = jetson_utils.videoSource('csi://0', argv=['--input-width=3840', '--input-height=2160', '--input-rate=10', '--input-flip=rotate-180']) 

Terminal log:

[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device csi://0
[gstreamer] gstCamera pipeline string:
[gstreamer] nvarguscamerasrc sensor-id=0 wbmode=3 tnr-mode=2 tnr-strength=1 ee-mode=2 ee-strength=1 exposurecompensation=0 ! video/x-raw(memory:NVMM), width=(int)3840, height=(int)2160, framerate=10/1, format=(string)NV12 ! nvvidconv flip-method=0 ! video/x-raw(memory:NVMM) ! appsink name=mysink
[gstreamer] gstCamera successfully created device csi://0
[video]  created gstCamera from csi://0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: csi://0
     - protocol:  csi
     - location:  0
  -- deviceType: csi
  -- ioType:     input
  -- width:      3840
  -- height:     2160
  -- frameRate:  10
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
------------------------------------------------
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> nvvconv0
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvvconv0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvvconv0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvarguscamerasrc0
[gstreamer] gstreamer message stream-start ==> pipeline0
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3840 x 2160 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 22.250000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 22.250000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 0 
   Output Stream W = 3840 H = 2160 
   seconds to Run    = 0 
   Frame Rate = 29.999999 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[gstreamer] gstCamera -- onPreroll
[gstreamer] gstBufferManager recieve caps:  video/x-raw(memory:NVMM), width=(int)3840, height=(int)2160, format=(string)NV12, framerate=(fraction)10/1
[gstreamer] gstBufferManager -- recieved first frame, codec=raw format=nv12 width=3840 height=2160 size=1008
[gstreamer] gstBufferManager -- recieved NVMM memory
[cuda]   allocated 4 ring buffers (8 bytes each, 32 bytes total)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink
[gstreamer] gstreamer message async-done ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> mysink
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline0
nvbuf_utils: dmabuf_fd 1256 mapped entry NOT found
nvbuf_utils: Can not get HW buffer from FD... Exiting...
NvBufferGetParams failed for dst_dmabuf_fd
nvbuffer_transform Failed
[cuda]   allocated 4 ring buffers (24883200 bytes each, 99532800 bytes total)

When flipping "horisontal" or "vertical" it works, it seems like only rotate-180 doesn't.

dusty-nv, 2023-05-01:
Hi @raulo987, due to this code, it's actually the default to use rotate-180 as most cameras need rotated:

https://github.com/dusty-nv/jetson-utils/blob/b29cf49639224351d55b66647e2adf89124626bf/camera/gstCamera.cpp#L156

Can you try either running it with `--input-flip=none` or commenting out that code block and recompiling/reinstalling?

paulblum, 2023-06-09:
I'm having the same crashing issue using 'rotate-180' with the CSI camera...

@dusty-nv I tried running with `--input-flip=none`, which didn't have an effect on the camera stream (gave the original upside-down output)

dusty-nv, 2023-06-09:
@paulblum can you provide your terminal log from when you run it?

paulblum, 2023-06-10:
@dusty-nv 

```
pb@jetson-nano:~$ cd jetson-inference/build/aarch64/bin
pb@jetson-nano:~/jetson-inference/build/aarch64/bin$ video-viewer 0 --input-flip=rotate-180
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device csi://0
[gstreamer] gstCamera pipeline string:
[gstreamer] nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=0 ! video/x-raw(memory:NVMM) ! appsink name=mysink
[gstreamer] gstCamera successfully created device csi://0
[video]  created gstCamera from csi://0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: csi://0
     - protocol:  csi
     - location:  0
  -- deviceType: csi
  -- ioType:     input
  -- width:      1280
  -- height:     720
  -- frameRate:  30
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[OpenGL] glDisplay -- display device initialized (1920x1080)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- width:      1920
  -- height:     1080
  -- frameRate:  0
  -- numBuffers: 4
  -- zeroCopy:   true
------------------------------------------------
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> nvvconv0
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvvconv0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvvconv0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvarguscamerasrc0
[gstreamer] gstreamer message stream-start ==> pipeline0
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;
[... 26 lines cut ...]
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline0
[cuda]   allocated 4 ring buffers (2764800 bytes each, 11059200 bytes total)
video-viewer:  captured 0 frames (1280x720)
[OpenGL] glDisplay -- set the window size to 1280x720
[OpenGL] creating 1280x720 texture (GL_RGB8 format, 2764800 bytes)
[cuda]   registered openGL texture for interop access (1280x720, GL_RGB8, 2764800 bytes)
video-viewer:  captured 1 frames (1280x720)
nvbuf_utils: dmabuf_fd 1052 mapped entry NOT found
nvbuf_utils: Can not get HW buffer from FD... Exiting...
NvBufferGetParams failed for dst_dmabuf_fd
nvbuffer_transform Failed
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
^Creceived SIGINT
[gstreamer] gstCamera::Capture() -- a timeout occurred waiting for the next image buffer
video-viewer:  shutting down...
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success
[gstreamer] gstCamera -- pipeline stopped
video-viewer:  shutdown complete
```
[terminal.log](https://github.com/dusty-nv/jetson-inference/files/11712830/terminal.log)

dusty-nv, 2023-06-12:
@paulblum please see my suggestion in https://github.com/dusty-nv/jetson-inference/issues/1673#issuecomment-1587825211