dusty-nv/jetson-inference #693: Failed TLS packet

TotalNoobLuke, 2020-08-24:
Sorry to be a bother but ive been searching everywhere to find a solution since thursday, ive checked all open and closed issues but did'nt find my problem.
when i try to clone or --recursive clone the /dusty-nv/jetson-inference.git 
it always bugs out at 68% and gives....

error: RPC failed; curl 56 GnuTLS recv error (-9): A TLS packet with unexpected length was received.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

i have however succeeded in cloning tokks repo but tht is 4yrs old now. 
i have tried installing to a clean install of the latest development kit too.
Also tried looking at logs but could'nt find the specific log in the /var/logs  directory.

Any idea what is causing this?

TotalNoobLuke, 2020-08-25:
Whatever you did its Fixed! Downloaded on first attempt today. Thank you very much

closed 2020-08-25 by TotalNoobLuke

dusty-nv, 2020-08-25:
@TotalNoobLuke I think it was a GitHub network issue - glad you were able to download it.

TotalNoobLuke, 2020-08-26:
Thanks for Replying Dusty! The inference pack is a game changer compared to what ive been learning so far, Great work!! Loving the Nano way more now. Thanks for dropping by Pauls show last week too was awesome to have someone frmom Nvidia come on. Have a nice day :)

dusty-nv, 2020-08-26:
Thanks for tuning in, that was a lot of fun!