dusty-nv/jetson-inference #694: Retrain SSD Mobilenet

tbo-bomarin, 2020-08-24:
Hi,

Really good toturial, even for noob like me!!

I am trying to make my own custom object detector based on a retrained SSD-Mobilenet. I have 401 jpegs with corresponding xmls.

To figure out how to save them, I took some pictures with the camera-capture app with detection setting enabled. And I "reverse engineerd" my dataset into that structure. Off course deleting the images, xml and file names in txt files left by my testing.

How should the files be divided to make it reasonable? Can I just list all file names in all txt files like the camera-capture app does with merge setting enabled?

dusty-nv, 2020-08-25:
> How should the files be divided to make it reasonable? Can I just list all file names in all txt files like the camera-capture app does with merge setting enabled?

Hi @tbo-bomarin, glad you are enjoying the tutorial!   Yes, in reality, with the limited number of images you have it should be just fine to replicate them all across the train/val/test datasets (like the 'merge' setting does).   You wouldn't want to do this for a production model, as the accuracy reported will include images it is already trained on.  So it's recommended to have some independent test set if possible (normally 80% train, 10% val, 10% test is fine)

tbo-bomarin, 2020-08-25:
Thanks for quick answer @dusty-nv!

I tried that, but I am getting an error still. Not quite sure where to start to find the fault.

Output from terminal:

tore@tore-desktop:~/jetson-inference/python/training/detection/ssd$ python3 train_ssd.py --dataset-type=voc --data=data/salmon --model-dir=models/salmon
2020-08-25 19:25:07 - Using CUDA...
2020-08-25 19:25:07 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/salmon', dataset_type='voc', datasets=['data/salmon'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=30, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2020-08-25 19:25:07 - Prepare training datasets.
2020-08-25 19:25:07 - VOC Labels read from file: ('BACKGROUND', 'salmon')
2020-08-25 19:25:07 - Stored labels into file models/salmon/labels.txt.
2020-08-25 19:25:07 - Train dataset size: 402
2020-08-25 19:25:07 - Prepare Validation datasets.
2020-08-25 19:25:08 - VOC Labels read from file: ('BACKGROUND', 'salmon')
2020-08-25 19:25:08 - Validation dataset size: 402
2020-08-25 19:25:08 - Build network.
2020-08-25 19:25:08 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
2020-08-25 19:25:08 - Took 0.50 seconds to load the model.
2020-08-25 19:25:27 - Learning rate: 0.01, Base net learning rate: 0.001, Extra Layers learning rate: 0.01.
2020-08-25 19:25:27 - Uses CosineAnnealingLR scheduler.
2020-08-25 19:25:27 - Start training from epoch 0.
/home/tore/.local/lib/python3.6/site-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
Traceback (most recent call last):
  File "train_ssd.py", line 343, in <module>
    device=DEVICE, debug_steps=args.debug_steps, epoch=epoch)
  File "train_ssd.py", line 113, in train
    for i, data in enumerate(loader):
  File "/home/tore/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 363, in __next__
    data = self._next_data()
  File "/home/tore/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 989, in _next_data
    return self._process_data(data)
  File "/home/tore/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 1014, in _process_data
    data.reraise()
  File "/home/tore/.local/lib/python3.6/site-packages/torch/_utils.py", line 395, in reraise
    raise self.exc_type(msg)
AttributeError: Caught AttributeError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/tore/.local/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 185, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/tore/.local/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/tore/.local/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/tore/.local/lib/python3.6/site-packages/torch/utils/data/dataset.py", line 207, in __getitem__
    return self.datasets[dataset_idx][sample_idx]
  File "/home/tore/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 58, in __getitem__
    boxes, labels, is_difficult = self._get_annotation(image_id)
  File "/home/tore/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 131, in _get_annotation
    is_difficult_str = object.find('difficult').text
AttributeError: 'NoneType' object has no attribute 'text'

dusty-nv, 2020-08-25:
It seems like either you have an image in the dataset that has no `<object>` defined in it's XML annotation, or it's missing the `<difficult>0</difficult>`  line inside of it's `<object>` tag.

Here is an example of what an XML annotation file can look like:

```xml
<annotation>
    <filename>20200713-154806.jpg</filename>
    <folder>drinks</folder>
    <source>
        <database>drinks</database>
        <annotation>custom</annotation>
        <image>custom</image>
    </source>
    <size>
        <width>1280</width>
        <height>720</height>
        <depth>3</depth>
    </size>
    <segmented>0</segmented>
    <object>
        <name>Coke</name>
        <pose>unspecified</pose>
        <truncated>0</truncated>
        <difficult>0</difficult>
        <bndbox>
            <xmin>480</xmin>
            <ymin>228</ymin>
            <xmax>707</xmax>
            <ymax>629</ymax>
        </bndbox>
    </object>
</annotation>
```

rom1vfr, 2020-08-27:
Hi, 
very nice tutorial ...but
I experienced the same issue, when try transfer learning tutorial and same error message, I checked the annotation and nothing in wront in the Pascal VOC annotation, any other idea?
when I do transfer learning with 1000image net and ONNX this works fine.
Thanks Rom1

dusty-nv, 2020-08-27:
Hi @rom1vfr @tbo-bomarin can you share your dataset with me so I can debug?   Thanks.

tbo-bomarin, 2020-08-27:
@dusty-nv. You were right with regards to the xmls on mine. I redid all labels with labelimg and then it worked.

I only trained 1 class and with a very limited and poor dataset. So it recognise salmon with almost 100% confidence. But when I poi t the camera at myself it is 87% sure that I am also a salmon 😂

closed 2023-04-27 by dusty-nv