dusty-nv/jetson-inference #696: retraining mobilenet

Yong-Hoon-In, 2020-08-27:
hi, thanks for supporting. I'm trying to retraining mobilenet ssd with my custom voc dataset. but now I'm face the error.
My dataset has 6949 jpg and 6949 xml file and 22 labels

this is when I start the training

~/jetson-inference/python/training/detection/pytorch-ssd$ python3 train_ssd.py --dataset-type voc --datasets /home/vision/objdetdata/ --num-epochs 32 --validation-epochs 1 --lr 0.01   --checkpoint-folder models/models_2 
2020-08-27 11:24:20 - Using CUDA...
2020-08-27 11:24:20 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/models_2', dataset_type='voc', datasets=['/home/vision/objdetdata/'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=32, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2020-08-27 11:24:20 - Prepare training datasets.
2020-08-27 11:24:20 - No labels file, using default VOC classes.
2020-08-27 11:24:20 - Stored labels into fil emodels/models_2/labels.txt.
2020-08-27 11:24:20 - Train dataset size: 6949
2020-08-27 11:24:20 - Prepare Validation datasets.
2020-08-27 11:24:20 - No labels file, using default VOC classes.
2020-08-27 11:24:20 - Validation dataset size: 6949
2020-08-27 11:24:20 - Build network.
2020-08-27 11:24:21 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
2020-08-27 11:24:21 - Took 0.58 seconds to load the model.
2020-08-27 11:24:36 - Learning rate: 0.01, Base net learning rate: 0.001, Extra Layers learning rate: 0.01.
2020-08-27 11:24:36 - Uses CosineAnnealingLR scheduler.
2020-08-27 11:24:36 - Start training from epoch 0.
/home/vision/.local/lib/python3.6/site-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)

here is the error what I face.

Traceback (most recent call last):
  File "train_ssd.py", line 345, in <module>
    device=DEVICE, debug_steps=args.debug_steps, epoch=epoch)
  File "train_ssd.py", line 113, in train
    for i, data in enumerate(loader):
  File "/home/vision/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 363, in __next__
    data = self._next_data()
  File "/home/vision/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 989, in _next_data
    return self._process_data(data)
  File "/home/vision/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 1014, in _process_data
    data.reraise()
  File "/home/vision/.local/lib/python3.6/site-packages/torch/_utils.py", line 395, in reraise
    raise self.exc_type(msg)
cv2.error: Caught error in DataLoader worker process 1.
Original Traceback (most recent call last):
  File "/home/vision/.local/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 185, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/vision/.local/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/vision/.local/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/vision/.local/lib/python3.6/site-packages/torch/utils/data/dataset.py", line 207, in __getitem__
    return self.datasets[dataset_idx][sample_idx]
  File "/home/vision/jetson-inference/python/training/detection/pytorch-ssd/vision/datasets/voc_dataset.py", line 85, in __getitem__
    image = self._read_image(image_id)
  File "/home/vision/jetson-inference/python/training/detection/pytorch-ssd/vision/datasets/voc_dataset.py", line 147, in _read_image
    image = cv2.cvtColor(image, cv2.COLOR_BGR2RGB)
cv2.error: OpenCV(4.1.1) /home/nvidia/host/build_opencv/nv_opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

dusty-nv, 2020-08-27:
It seems like one of your images may be corrupt and openCV can't read it?

You might want to add additional logging to `voc_dataset.py` to find out which image it is.

Yong-Hoon-In, 2020-08-27:
> It seems like one of your images may be corrupt and openCV can't read it?
> 
> You might want to add additional logging to `voc_dataset.py` to find out which image it is.

I'm really thanks for the quick reply I'll check my data

closed 2023-04-27 by dusty-nv