dusty-nv/jetson-inference #698: pip3 install -v -r requirements.txt failed on Xavier

DennisFaucher, 2020-08-27:
Trying to install TF training on my Xavier using your instructions. JetPack version Version: 4.4-b186. Failed with this error.

 Running setup.py bdist_wheel for numpy ...   Destination directory: /tmp/tmpbpwhko8spip-wheel-
  Running command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-kjp3gpkz/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpbpwhko8spip-wheel- --python-tag cp36
  Running from numpy source directory.
  Cythonizing sources
  Processing numpy/random/_bounded_integers.pxd.in
  Processing numpy/random/_mt19937.pyx
  Traceback (most recent call last):
    File "/tmp/pip-build-kjp3gpkz/numpy/tools/cythonize.py", line 235, in <module>
      main()
    File "/tmp/pip-build-kjp3gpkz/numpy/tools/cythonize.py", line 231, in main
      find_process_files(root_dir)
    File "/tmp/pip-build-kjp3gpkz/numpy/tools/cythonize.py", line 222, in find_process_files
      process(root_dir, fromfile, tofile, function, hash_db)
    File "/tmp/pip-build-kjp3gpkz/numpy/tools/cythonize.py", line 188, in process
      processor_function(fromfile, tofile)
    File "/tmp/pip-build-kjp3gpkz/numpy/tools/cythonize.py", line 76, in process_pyx
      raise RuntimeError(f'Building {VENDOR} requires Cython >= {required_version}')
  RuntimeError: Building NumPy requires Cython >= 0.29.21
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-build-kjp3gpkz/numpy/setup.py", line 499, in <module>
      setup_package()
    File "/tmp/pip-build-kjp3gpkz/numpy/setup.py", line 479, in setup_package
      generate_cython()
    File "/tmp/pip-build-kjp3gpkz/numpy/setup.py", line 274, in generate_cython
      raise RuntimeError("Running cythonize failed!")
  RuntimeError: Running cythonize failed!
error
  Failed building wheel for numpy
  Running setup.py clean for numpy
  Running command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-kjp3gpkz/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" clean --all
  Running from numpy source directory.

  `setup.py clean` is not supported, use one of the following instead:

    - `git clean -xdf` (cleans all files)
    - `git clean -Xdf` (cleans all versioned files, doesn't touch
                        files that aren't checked into the git repo)

  Add `--force` to your command to use it anyway if you must (unsupported).

  Failed cleaning build dir for numpy
Successfully built pandas
Failed to build numpy
Installing collected packages: docutils, jmespath, urllib3, six, python-dateutil, botocore, s3transfer, boto3, pytz, numpy, pandas

  Running setup.py install for numpy ...     Running command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-kjp3gpkz/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-lg82v95s-record/install-record.txt --single-version-externally-managed --compile --user --prefix=
    Running from numpy source directory.

    Note: if you need reliable uninstall behavior, then install
    with pip instead of using `setup.py install`:

      - `pip install .`       (from a git repo or downloaded source
                               release)
      - `pip install numpy`   (last NumPy release on PyPi)

    Cythonizing sources
    numpy/random/_bounded_integers.pxd.in has not changed
    Processing numpy/random/_mt19937.pyx
    Traceback (most recent call last):
      File "/tmp/pip-build-kjp3gpkz/numpy/tools/cythonize.py", line 235, in <module>
        main()
      File "/tmp/pip-build-kjp3gpkz/numpy/tools/cythonize.py", line 231, in main
        find_process_files(root_dir)
      File "/tmp/pip-build-kjp3gpkz/numpy/tools/cythonize.py", line 222, in find_process_files
        process(root_dir, fromfile, tofile, function, hash_db)
      File "/tmp/pip-build-kjp3gpkz/numpy/tools/cythonize.py", line 188, in process
        processor_function(fromfile, tofile)
      File "/tmp/pip-build-kjp3gpkz/numpy/tools/cythonize.py", line 76, in process_pyx
        raise RuntimeError(f'Building {VENDOR} requires Cython >= {required_version}')
    RuntimeError: Building NumPy requires Cython >= 0.29.21
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-kjp3gpkz/numpy/setup.py", line 499, in <module>
        setup_package()
      File "/tmp/pip-build-kjp3gpkz/numpy/setup.py", line 479, in setup_package
        generate_cython()
      File "/tmp/pip-build-kjp3gpkz/numpy/setup.py", line 274, in generate_cython
        raise RuntimeError("Running cythonize failed!")
    RuntimeError: Running cythonize failed!
error
Cleaning up...
  Removing source in /tmp/pip-build-kjp3gpkz/numpy
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-kjp3gpkz/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-lg82v95s-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-kjp3gpkz/numpy/
Exception information:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 360, in run
    prefix=options.prefix_path,
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 784, in install
    **kwargs
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 878, in install
    spinner=spinner,
  File "/usr/lib/python3/dist-packages/pip/utils/__init__.py", line 725, in call_subprocess
    % (command_desc, proc.returncode, cwd))
pip.exceptions.InstallationError: Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-kjp3gpkz/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-lg82v95s-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-kjp3gpkz/numpy/

dusty-nv, 2020-08-27:
Hi @DennisFaucher , which container and tag are you running?

```
RuntimeError: Building NumPy requires Cython >= 0.29.21
```

Could you try running `pip3 install --upgrade cython` in the container first?

DennisFaucher, 2020-08-27:
I'm using native JetPack on Xavier. All the packages load in Python, so I'm ignoring the error for now and training a new model on the Xavier.

dusty-nv, 2020-08-27:
OK gotcha.  Since this is unrelated to the containers, if you have issues in the future using TF natively on Jetson, please post to one of the Jetson forums:

https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/70

closed 2020-08-27 by dusty-nv