dusty-nv/jetson-inference #703: several boxes around same object

linukc, 2020-08-31:
Hi. I create custom detectnet model (python) with my test dataset (60 photo). Is this bug(photo)
 linked with small dataset size? Have we got a instrument for avoiding boxes overlay or i have to wright my own? I don't want use box with highest score, because several object can be on the scene at the same time.
[image]

berkantay, 2020-10-06:
What is the number of classes? After the transfer learning is done script mentions the number of classes according to labeling? It might be because of multipe classes pointing same object.

linukc, 2020-10-11:
I have only one class in my model.

linukc, 2020-10-18:
okay, all what i needed was method for non-maximum suppression.

closed 2020-10-18 by linukc