dusty-nv/jetson-inference #706: cannot successfully run jetson nano imagenet commands

xaseven, 2020-09-03:
I am trying to run the jetson-inference samples on jetson nano. Running the imagenet  test app results in the following error:
command: $ ./imagenet --network=coco-dog ./images/dog_0.jpg out_dog_0.jpg

ERRINFO:
 -- URI: file:///home/nvidia/jetson-inference/build/aarch64/bin/./images/dog_0.jpg
     - protocol:  file
     - location:  ./images/dog_0.jpg
     - extension: jpg
  -- deviceType: file
  -- ioType:     input
  -- codec:      unknown
  -- width:      0
  -- height:     0
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[video]  created imageWriter from file:///home/nvidia/jetson-inference/build/aarch64/bin/out_dog_0.jpg
------------------------------------------------
imageWriter video options:
------------------------------------------------
  -- URI: file:///home/nvidia/jetson-inference/build/aarch64/bin/out_dog_0.jpg
     - protocol:  file
     - location:  out_dog_0.jpg
     - extension: jpg
  -- deviceType: file
  -- ioType:     output
  -- codec:      unknown
  -- width:      0
  -- height:     0
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1680x1050
[OpenGL] glDisplay -- X window resolution:    1680x1050
[OpenGL] glDisplay -- display device initialized (1680x1050)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1680
  -- height:     1050
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[TRT]    imageNet -- failed to initialize.
imagenet:  failed to initialize imageNet

my networks dir:
~/jetson-inference/build/aarch64/bin/networks$ ls
DetectNet-COCO-Dog  detectnet.prototxt  ilsvrc12_synset_words.txt  ssd_coco_labels.txt

my DetectNet-COCO-Dog :
~/jetson-inference/build/aarch64/bin/networks/DetectNet-COCO-Dog$ ls
class_labels.txt  digits.deploy.prototxt  mean.binaryproto   snapshot_iter_38600.caffemodel                           solver.prototxt
deploy.prototxt   info.json               original.prototxt  snapshot_iter_38600.caffemodel.1.1.7103.GPU.FP16.engine  train_val.prototxt

Can someone please suggest me what all things i can solve this issues

dusty-nv, 2020-09-03:
Hi there, I think you mean to use detectnet sample with coco-dog model.  That is an object detection model, not an image classification model.

xaseven, 2020-09-03:
yes,now i use command like :
python3 detectnet.py --network=coco-dog imges/test1.mp4 out_1.mp4
is ok....
However, dogs can be identified in the video file, sometimes even people are identified; do I need to train the model again?

dusty-nv, 2020-09-03:
I would recommend using the `ssd-mobilenet-v2`.  `coco-dog` is based on an outdated/slow DNN architecture.