dusty-nv/jetson-inference #710: Using gstreamer as an input to the camera demos?

Tetsujinfr, 2020-09-08:
Hi, is it possible to feed the camera demos, segnet-camera.py with a gstreamer string, instead of the v4l2 camera device option?
I need to setup my camera in a specific mode, which by default is not the one v4l2 grabs.

I would like to avoid touching the code and do this on the commamd line but if I need to I can.

Thanks

dusty-nv, 2020-09-08:
Hi @Tetsujinfr , it doesn't accept a general GStreamer string as there are some dynamic things happening with the pipeline generation, such as the detect codec and desired format/resolution.

You can add `v4l2src` options in the code here:  https://github.com/dusty-nv/jetson-utils/blob/cae2730a09edd955631fcbdfc8c4c38556a93600/camera/gstCamera.cpp#L154

After you have made edits, be sure to re-run `make` and `sudo make install`

Tetsujinfr, 2020-09-08:
Ia there a way I can specify the resolution of the v4l2 camera at command line level?
Also, I have a stereo camera, hence I would like to pass only the left view through a gstreamer command, I am not sure I can do that with v4l2.

Tetsujinfr, 2020-09-08:
Btw, I can see that gstreamer is used in the pipeline, so I was thinking, worst case scenario I can hardode my gstreamer command at the right place in the code no?

dusty-nv, 2020-09-08:
> Ia there a way I can specify the resolution of the v4l2 camera at command line level?

Yes, you can use the `--input-width` and `--input-height` arguments.  Run the program with `--help` or see here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#input-options

> Also, I have a stereo camera, hence I would like to pass only the left view through a gstreamer command, I am not sure I can do that with v4l2.
It depends on if the stereo camera is presented as one stream (i.e. one v4l2 device) or multiple streams (multiple v4l2 devices).

For example, Stereolabs ZED has both left/right images in one frame, so then you need to crop the image if you only want the left view.  See [`cuda-examples.py`](https://github.com/dusty-nv/jetson-utils/blob/cae2730a09edd955631fcbdfc8c4c38556a93600/python/examples/cuda-examples.py) for an example of using the `cudaCrop()` function from Python, and from this page:  

https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#cropping

> Btw, I can see that gstreamer is used in the pipeline, so I was thinking, worst case scenario I can hardode my gstreamer command at the right place in the code no?
Yes you could do that, the point in the code would be here:   https://github.com/dusty-nv/jetson-utils/blob/cae2730a09edd955631fcbdfc8c4c38556a93600/camera/gstCamera.cpp#L154

Tetsujinfr, 2020-09-20:
ok thanks for your reply.

modifying the jetson utils source code is too complex for me I am afraid.
I am playing with the segnet.py python example, just need to get a gstreamer image capture into the input_img object properly [here](https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/segnet.py#71).
I am using cv2 to get the gstreamer frame of my stereo camera properly (it works this way).

However the input_img object is a "cudaImage": I am not familiar with this object.
cv2 returns a "numpy.ndarray" frame. How can I convert that into a cudaImage?
I see there is a cudaToNumpy function in your jetson.utils lib but I did not find the reverse function, 

Btw, any chance you consider adding the option to feed directly a gstreamer string into your lib videoSource code base? would give more flexibility no?

Thanks for your help on this

liboning1996, 2022-08-16:
> ok thanks for your reply.
> 
> modifying the jetson utils source code is too complex for me I am afraid. I am playing with the segnet.py python example, just need to get a gstreamer image capture into the input_img object properly [here](https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/segnet.py#71). I am using cv2 to get the gstreamer frame of my stereo camera properly (it works this way), and I am simply trying to inject that into your code as an input image, rather than using the input = jetson.utils.videoSource() and input.Capture() methods.
> 
> However the input_img object is a "cudaImage": I am not familiar with this object. cv2 returns a "numpy.ndarray" frame. How can I convert that into a cudaImage? I see there is a cudaToNumpy function in your jetson.utils lib but I did not find the reverse function,
> 
> Btw, any chance you consider adding the option to feed directly a gstreamer string into your lib videoSource code base? would give more flexibility no?
> 
> Thanks for your help on this

hello,
I meet the same question with you, have you solve this question? could you plz show your solution ,thank you very very very much!

closed 2023-04-27 by dusty-nv