dusty-nv/jetson-inference #712: Is it possible to train with rtx 3080 graphics card? I wonder if it fits the architecture that rtx introduces.

ingbeeedd, 2020-09-08:
please answer my question..

dusty-nv, 2020-09-08:
Yes, it can run CUDA/cuDNN/ect, and the pytorch training codes run on Linux PC as well.  you will need to install PyTorch following pytorch.org, not the jetson installer on PC.

Also you do not need to clone the entire jetson-inference repo on PC, you can just clone my pytorch submodules:

* https://github.com/dusty-nv/pytorch-classification
* https://github.com/dusty-nv/pytorch-ssd

ingbeeedd, 2020-09-08:
Is the content that you mentioned only applied to rtx 30 or rtx series?

dusty-nv, 2020-09-08:
It would be essentially any CUDA-capable GPU, so any of the relatively recent ones.  I personally use a 1070 for training in my laptop.   If it has >= 8GB VRAM that is nice.   Probably I wouldn't go less than 6GB VRAM for a training GPU.

closed 2020-09-09 by ingbeeedd