dusty-nv/jetson-inference #714: How can I build the inference and utils modules for pyenv

maxsmadt, 2020-09-09:
Hi,

I am using pyenv to manage my python versions (and also manually built OpenCV with CUDA support). I would like to ask how I can correctly build the jetson inference and utils modules for those environments?

LewsTherin511, 2021-11-08:
Did you ever solve this? I'm having the same issue (using conda environment, but that shouldn't be a problem).
I ran the whole building procedure successfully, and I can use detectnet from the system python installation, but when switching to another virtual environment I get the error

    "ModuleNotFoundError: No module named 'jetson'"

dusty-nv, 2021-11-08:
Let's continue this topic in the new thread here:  https://github.com/dusty-nv/jetson-inference/issues/1285

closed 2021-11-08 by dusty-nv