dusty-nv/jetson-inference #715: Retraining on custom VOC dataset only classifies single object

JaredP94, 2020-09-09:
Hi @dusty-nv.

I've put together a collection of images, than contain a few objects within an image that are to be classified. I've used LabelImg to draw bounding boxes over each object of interest and generate the Pascal VOC annotations.

I re-train the model, which outputs the expected number of labels in the model/custom_model directory. However, when I run the model on a live camera feed, seemingly only the first entry in the labels list is what gets bounded and shown on the live feed (with good accuracy). The others never appear. Does this perhaps relate to how the scripts parse the annotations?

Example of a generated annotation below:
```
<annotation>
	<folder>JPEGImages</folder>
	<filename>20200908-224946.jpg</filename>
	<path>data</path>
	<source>
		<database>Unknown</database>
	</source>
	<size>
		<width>1280</width>
		<height>720</height>
		<depth>3</depth>
	</size>
	<segmented>0</segmented>
	<object>
		<name>modem</name>
		<pose>Unspecified</pose>
		<truncated>0</truncated>
		<difficult>0</difficult>
		<bndbox>
			<xmin>374</xmin>
			<ymin>314</ymin>
			<xmax>673</xmax>
			<ymax>516</ymax>
		</bndbox>
	</object>
	<object>
		<name>m_ant</name>
		<pose>Unspecified</pose>
		<truncated>0</truncated>
		<difficult>0</difficult>
		<bndbox>
			<xmin>315</xmin>
			<ymin>455</ymin>
			<xmax>363</xmax>
			<ymax>493</ymax>
		</bndbox>
	</object>
	<object>
		<name>gps_ant</name>
		<pose>Unspecified</pose>
		<truncated>0</truncated>
		<difficult>0</difficult>
		<bndbox>
			<xmin>1095</xmin>
			<ymin>249</ymin>
			<xmax>1148</xmax>
			<ymax>297</ymax>
		</bndbox>
	</object>
</annotation>
```

Please let me know if you require any additional information. 

Thanks in advance!

silent-code, 2020-09-14:
@JaredP94 not trying to hijack your question, but can you tell me how you got the voc dataset into a format so that the train_ssd.py script can function properly? I have used the labelImg tool as well but the output by itself does not appear to be in the appropriate format. I get the error: FileNotFoundError: [Errno 2] No such file or directory: 'data-voc/ImageSets/Main/trainval.txt' when running train_ssd.py. What subsequent steps with the voc_dataset.py script did you apply? And where did you get this script? Thanks in advance, Tim

JaredP94, 2020-09-15:
Hi @dusty-nv, just following up on this :)

dusty-nv, 2020-09-15:
Hi @JaredP94 , how many annotations does each of your classes have?   Are they fairly even?

JaredP94, 2020-09-15:
Hi @dusty-nv, I currently have four different classes configured with roughly 100, 80, 70, 60 annotations for each respective class (not a lot at the moment but there's minimal variance in train and test conditions to initially validate the model for multiple object detection). How closely matched should the number of classes be? The objects are classified with shared annotations (i.e one annotation may contain all four classes) - I don't imagine that would be an issue?

dusty-nv, 2020-09-16:
I'm not sure if that's too few, or perhaps unbalanced.   Also having multiple classes per image does ratchet up the complexity.   As a proof of concept, you might just want to keep it simple and interate from there.

The other thing you could do is try the eval_ssd.py script to see if the trained pytorch model is working, or if something is going on during deployment.

When you run the model with detectnet program, you are using the labels file that was saved with the model, correct?   It's first class should be BACKGROUND - it gets added during training.  The labels.txt from your dataset wouldn't have it, but the model is trained with it.

JaredP94, 2020-09-19:
Hi @dusty-nv, thanks for the feedback :) 

I'll start by increasing the dataset size, followed by separating the annotations to have a single class per annotation for simplicity. I'll report back once I've had a chance to do this and evaluate the results.

I ran the **eval_ssd.py** script and yielded the following:
```
Average Precision Per-class:
modem: 1.0000000000000002
m_ant: 0.018233526218269926
gps_ant: 0.33333333333333326
mcu: 1.0000000000000002
```

Based on those results, I would certainly expect to see at least the **modem** and **mcu** classes identified when running the camera feed, opposed to only the modem.

Regarding the labels file, I'm using the generated file that has the BACKGROUND entry inserted on the first line.

Thanks again for looking into this!

JaredP94, 2020-10-02:
Hi @dusty-nv,

Finally got around to testing with dataset changes. I decided to increase the size of the dataset rather than change how the annotations are structured since the requirement for multiple bounding boxes per image is essential in the use case.

I ran the **eval_ssd.py** script and yielded the following:
```
Average Precision Per-class:
modem: 1.0000000000000002
m_ant: 0.4764685786786765
gps_ant: 0.272875748654866
mcu: 1.0000000000000002
```

I still only observed a single bounding box output when running image analysis (both live and static). I decided to test out using the mb2-ssd-lite model instead which yielded closely matching evaluation results, however did render multiple bounding boxes on the image analysis (both live and static).

No real theories on why mb1-ssd isn't happy but achieving what I need now. Happy to look into anything if you're interested but otherwise I'll close this issue.

Thanks again for your assistance :)

closed 2020-10-02 by JaredP94